wasm-micro-runtime/wasm-micro-runtime #1758: Unable to build lldb on MacOS M1

eloparco, 2022-11-27:
Hello,
When I try to build `lldb` as described [here](https://github.com/bytecodealliance/wasm-micro-runtime/blob/12bbb9189cfb0f9094f7f60d0b69ece48cd46a46/doc/source_debugging.md#debugging-with-aot) I get the following error:
```
[2/1197] Building CXX object tools/lldb/source/Core/CMakeFiles/lldbCore.dir/DumpDataExtractor.cpp.o
FAILED: tools/lldb/source/Core/CMakeFiles/lldbCore.dir/DumpDataExtractor.cpp.o
/Library/Developer/CommandLineTools/usr/bin/c++ -DGTEST_HAS_RTTI=0 -DHAVE_ROUND -DLLDB_CONFIGURATION_DEBUG -D_DEBUG -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/build/tools/lldb/source/Core -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/lldb/source/Core -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/build/tools/lldb/source -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/lldb/include -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/build/tools/lldb/include -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/build/include -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/llvm/include -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/llvm/../clang/include -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/build/tools/lldb/../clang/include -I/opt/homebrew/include -I/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/include/libxml2 -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/lldb/source/. -I/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/include -fPIC -fvisibility-inlines-hidden -Werror=date-time -Werror=unguarded-availability-new -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -pedantic -Wno-long-long -Wc++98-compat-extra-semi -Wimplicit-fallthrough -Wcovered-switch-default -Wno-noexcept-type -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wsuggest-override -Wstring-conversion -Wmisleading-indentation -fdiagnostics-color -Wno-deprecated-declarations -Wno-unknown-pragmas -Wno-strict-aliasing -Wno-deprecated-register -Wno-vla-extension -g -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk -mmacosx-version-min=12.6  -fno-exceptions -fno-rtti -std=c++14 -MD -MT tools/lldb/source/Core/CMakeFiles/lldbCore.dir/DumpDataExtractor.cpp.o -MF tools/lldb/source/Core/CMakeFiles/lldbCore.dir/DumpDataExtractor.cpp.o.d -o tools/lldb/source/Core/CMakeFiles/lldbCore.dir/DumpDataExtractor.cpp.o -c /Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/lldb/source/Core/DumpDataExtractor.cpp
/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/lldb/source/Core/DumpDataExtractor.cpp:241:12: error: expected unqualified-id
  if (std::isnan(f)) {
           ^
/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/include/math.h:165:5: note: expanded from macro 'isnan'
    ( sizeof(x) == sizeof(float)  ? __inline_isnanf((float)(x))          \
    ^
/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/lldb/source/Core/DumpDataExtractor.cpp:242:14: error: expected unqualified-id
    if (std::signbit(f))
             ^
/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/include/math.h:170:5: note: expanded from macro 'signbit'
    ( sizeof(x) == sizeof(float)  ? __inline_signbitf((float)(x))        \
    ^
/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/lldb/source/Core/DumpDataExtractor.cpp:247:12: error: expected unqualified-id
  if (std::isinf(f)) {
           ^
/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/include/math.h:160:5: note: expanded from macro 'isinf'
    ( sizeof(x) == sizeof(float)  ? __inline_isinff((float)(x))          \
    ^
/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/lldb/source/Core/DumpDataExtractor.cpp:248:14: error: expected unqualified-id
    if (std::signbit(f))
             ^
/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/include/math.h:170:5: note: expanded from macro 'signbit'
    ( sizeof(x) == sizeof(float)  ? __inline_signbitf((float)(x))        \
    ^
4 errors generated.
[13/1197] Building CXX object tools/lldb/source/DataFormatters/CMakeFiles/lldbDataFormatters.dir/TypeSummary.cpp.o
ninja: build stopped: subcommand failed.
```
This happens on my MacOS M1 environment.
Any idea on how to solve it?
Thanks

eloparco, 2022-11-27:
Update: if I just remove the `std::` prefix for the functions generating the error (for instance `std::isnan(f)`), it temporarily solves the problem, but then I bump into a new one:
```[3/1060] Building CXX object tools/lldb/source/Plugins/ExpressionParser/Clang/CMakeFiles/lldbPluginExpressionParserClang.dir/IRDynamicChecks.cpp.o
FAILED: tools/lldb/source/Plugins/ExpressionParser/Clang/CMakeFiles/lldbPluginExpressionParserClang.dir/IRDynamicChecks.cpp.o
/Library/Developer/CommandLineTools/usr/bin/c++ -DGTEST_HAS_RTTI=0 -DHAVE_ROUND -DLLDB_CONFIGURATION_DEBUG -D_DEBUG -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/build/tools/lldb/source/Plugins/ExpressionParser/Clang -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/lldb/source/Plugins/ExpressionParser/Clang -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/build/tools/lldb/source -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/lldb/include -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/build/tools/lldb/include -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/build/include -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/llvm/include -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/llvm/../clang/include -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/build/tools/lldb/../clang/include -I/opt/homebrew/include -I/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/include/libxml2 -I/Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/lldb/source/. -I/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/include -fPIC -fvisibility-inlines-hidden -Werror=date-time -Werror=unguarded-availability-new -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -pedantic -Wno-long-long -Wc++98-compat-extra-semi -Wimplicit-fallthrough -Wcovered-switch-default -Wno-noexcept-type -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wsuggest-override -Wstring-conversion -Wmisleading-indentation -fdiagnostics-color -Wno-deprecated-declarations -Wno-unknown-pragmas -Wno-strict-aliasing -Wno-deprecated-register -Wno-vla-extension -g -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk -mmacosx-version-min=12.6  -fno-exceptions -fno-rtti -std=c++14 -MD -MT tools/lldb/source/Plugins/ExpressionParser/Clang/CMakeFiles/lldbPluginExpressionParserClang.dir/IRDynamicChecks.cpp.o -MF tools/lldb/source/Plugins/ExpressionParser/Clang/CMakeFiles/lldbPluginExpressionParserClang.dir/IRDynamicChecks.cpp.o.d -o tools/lldb/source/Plugins/ExpressionParser/Clang/CMakeFiles/lldbPluginExpressionParserClang.dir/IRDynamicChecks.cpp.o -c /Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/lldb/source/Plugins/ExpressionParser/Clang/IRDynamicChecks.cpp
In file included from /Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/lldb/source/Plugins/ExpressionParser/Clang/IRDynamicChecks.cpp:9:
In file included from /Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/build/include/llvm/IR/Constants.h:23:
In file included from /Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/build/include/llvm/ADT/APFloat.h:19:
In file included from /Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/build/include/llvm/ADT/APInt.h:19:
In file included from /Users/eloparco/dev/wasm-micro-runtime/core/deps/llvm/build/include/llvm/Support/MathExtras.h:19:
/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/usr/include/c++/v1/cmath:642:26: error: no template named 'numeric_limits'
    bool _FloatBigger = (numeric_limits<_FloatT>::digits > numeric_limits<_IntT>::digits),
                         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/usr/include/c++/v1/cmath:642:60: error: no template named 'numeric_limits'
    bool _FloatBigger = (numeric_limits<_FloatT>::digits > numeric_limits<_IntT>::digits),
                                                           ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/usr/include/c++/v1/cmath:643:18: error: no template named 'numeric_limits'
    int _Bits = (numeric_limits<_IntT>::digits - numeric_limits<_FloatT>::digits)>
                 ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/usr/include/c++/v1/cmath:643:50: error: no template named 'numeric_limits'
    int _Bits = (numeric_limits<_IntT>::digits - numeric_limits<_FloatT>::digits)>
                                                 ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/usr/include/c++/v1/cmath:648:17: error: no template named 'numeric_limits'
  static_assert(numeric_limits<_FloatT>::radix == 2, "FloatT has incorrect radix");
                ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/usr/include/c++/v1/cmath:651:25: error: no template named 'numeric_limits'
  return _FloatBigger ? numeric_limits<_IntT>::max() :  (numeric_limits<_IntT>::max() >> _Bits << _Bits);
                        ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/usr/include/c++/v1/cmath:651:58: error: no template named 'numeric_limits'
  return _FloatBigger ? numeric_limits<_IntT>::max() :  (numeric_limits<_IntT>::max() >> _Bits << _Bits);
                                                         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/usr/include/c++/v1/cmath:661:16: error: no template named 'numeric_limits'
  using _Lim = numeric_limits<_IntT>;
               ^
[... cut ...]
/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/usr/include/c++/v1/cmath:665:21: error: use of undeclared identifier '_Lim'
  } else if (__r <= _Lim::lowest()) {
                    ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/usr/include/c++/v1/cmath:666:12: error: use of undeclared identifier '_Lim'
    return _Lim::min();
           ^
11 errors generated.
[14/1060] Building CXX object tools/lldb/source/Plugins/Language/CPlusPlus/CMakeFiles/lldbPluginCPlusPlusLanguage.dir/BlockPointer.cpp.o
ninja: build stopped: subcommand failed.

loganek, 2022-11-27:
Hi @eloparco; I think it's not an issue with WAMR, but rather with LLVM. However, I looked into that briefly, and I think the problem is with the way CMake generates compilation command and set of include directives, especially those two look suspicious to me:
```
 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk
-I/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/include
```
as I 1) wouldn't expect the sdk to be included twice, 2) I'd expect those paths to point to the same SDK. I'm mainly developing on Linux but also have MacOS M1 so I can try to reproduce and debug it tomorrow.

eloparco, 2022-11-27:
I just fixed it by deleting one of the SDK versions from `/Library/Developer/CommandLineTools/SDKs`, to avoid headers being picked from different versions. 
This is an issue already raised on the llvm project repo: https://github.com/llvm/llvm-project/issues/54988.

closed 2022-11-27 by eloparco