wasm-micro-runtime/wasm-micro-runtime #1764: Recommendation for function imports with wasm_c_api

nchamzn, 2022-11-28:
Hi,

I've been looking in to the ``wasm_c_api`` for function imports, I would like to use this API so that I can link native functions at instantiate time. After experimenting with this API and adding function imports [like in the example](https://github.com/WebAssembly/wasm-c-api/blob/main/example/callback.c#L116) I found that because my module is using Wasi that I need to link all Wasi functions as well.

Is there a recommended approach for using this API when using Wasi?  I couldn't find a way to get the list of Wasi functions in order to append to my imports.

Also, it also looks like this API is only going to work with native functions which are registered with ``wasm_runtime_register_natives`` - is that correct?

lum1n0us, 2022-12-01:
By default, (`-DWAMR_BUILD_LIBC_WASI=1`), there are built-in WASI support native functions in WAMR (iwasm). In theory, there is no necessary to link with user own WASI implementations unless they are unimplemented WASI APIs or user-customized implementation. 

`wasm_runtime_register_natives` is another way to setup host functions beside Wasm C APIs when programming with *wasm_export.h*.

nchamzn, 2022-12-01:
Sorry if my question wasn't clear. I'm creating a module instance with `wasm_instance_new_with_args`, and hoping to pass `imports` to this and link native functions as the module is instantiated, [like here in the example](https://github.com/WebAssembly/wasm-c-api/blob/main/example/callback.c#L102-L116).  However, when the module is built with `wasm32-wasi` and imports Wasi functions, instantiate will fail because the Wasi functions are not in my imports vector.

Is there any way to successfully instantiate a module when passing imports to `wasm_instance_new_with_args` like this?

yamt, 2022-12-12:
> Hi,
> 
> I've been looking in to the `wasm_c_api` for function imports, I would like to use this API so that I can link my native functions at instantiate time. After experimenting with this API and adding function imports [like in the example](https://github.com/WebAssembly/wasm-c-api/blob/main/example/callback.c#L116) I found that because my module is using Wasi that I need to link all Wasi functions as well.
> 
> Is there a recommended approach for using this API when using Wasi? I couldn't find a way to get the list of Wasi functions in order to append to my imports.
> 
> Also, it also looks like this API is only going to work with native functions which are registered with `wasm_runtime_register_natives` - is that correct?

do you mean you want to have a module which uses wasi
and you also want to make it import some extra functions at instantiate time?
unfortunately, wamr doesn't provide any reasonable way to do it as far as i understand.

nchamzn, 2023-01-17:
Hi @lum1n0us 

I just updated Wamr to head of main and this approach no longer works, I get this assertion failure:

```
libs/wamr/wamr/core/iwasm/common/wasm_c_api.c, line 5009, wasm_instance_new_with_args failed: Failed to match imports
```

I believed caused by #1843?

lum1n0us, 2023-01-19:
Thanks. I saw the point. It is a trick task to check and link `imports`.

*wasm_loader* will provide built-in host function supporting during loading phase. like `wasi_snapshot_preview1.fd_seek` when `WAMR_BUILD_LIBC_WASI=1`. Assuming caller will provide all necessary imports will be not appropriate in some scenarios.

Using `imports` as a supply of built-in seems to be a right way for now. But it requires users(caller) to 
1) try-run the target Wasm module via `iwasm` and collect warnings like *warning: failed to link import ...*. Those failed items require users to fill with `imports`
2) traverse `wasm_importtype_vec_t` from `wasm_module_imports()` to make sure the order of required link items by names from `wasm_importtype_name()`
3) build a proper `imports` and pass it with `wasm_instance_new`.

yamt, 2023-01-19:
> Thanks. I saw the point. It is a trick task to check and link `imports`.
>
> _wasm_loader_ will provide built-in host function supporting during loading phase. like `wasi_snapshot_preview1.fd_seek` [...]

there is no proper way to fill the wasm_importtype_vec_t slots for eg. wasi functions, is it?

it's an impedance mismatch between wasm and wamr: https://github.com/bytecodealliance/wasm-micro-runtime/issues/1353
while we can band-aid fix (like having a convenient function to fill wasm_importtype_vec_t using some special values to represent the `pre-linked` state)
a cleaner/longer-term fix might involve a major surgery.

lum1n0us, 2023-01-19:
#1902 is a workaround by skipping those items linked by loader

closed 2023-04-27 by wenyongh