wasm-micro-runtime/wasm-micro-runtime #1771: File I/O doesn't work inside a sub-module (-mexec-module=reactor)

blackbird1, 2022-11-30:
I've been using MULTI_MODULE support and it appears for whatever reason file I/O no longer works when setting a wasm module with the following:
target_link_options(file PRIVATE -mexec-model=reactor)

I need to be able to create sub-modules and it seems I'm only successful with -mexec-model=reactor. If I don't use that flag I get this this when trying to load as a sub-module "WASM module load failed: a command (with _start function) can not be a sub-module".

lum1n0us, 2022-12-01:
May I take a look at the source code?

blackbird1, 2022-12-10:
Actually basic I/O doesn't work at all if you use -mexec-module=reactor

`
    add_executable(file file.cpp)
    target_link_options(file PRIVATE -mexec-model=reactor)
`

`
#include <cstdio>
#include <cstring>
#include <cstdlib>
#include <unistd.h>
#include <fcntl.h>
#include <cerrno>

extern "C" {

__attribute__((export_name("read_file"))) size_t read_file(const char *path, char *buf, int bufSize) {
    printf("read_file: path=%s, bufSize=%d\n", path, bufSize);
    auto in = open(path, O_RDONLY);
    printf("read_file: open for path %s, errno=%d\n", path, errno);
    if(in == -1){
        fprintf(stderr, "read_file: open failed for path %s, errno=%d\n", path, errno);
        return 0;
    }
    auto bytesRead = read(in, buf, bufSize);
    close(in);
    return bytesRead;
}

__attribute__((export_name("write_file"))) int write_file(char *data, int dataSize, char *path) {

    printf("write_file: called for path %s data size=%d\n", path, dataSize);
    auto fd = open(path, O_WRONLY | O_CREAT | O_TRUNC, S_IRUSR | S_IWUSR);
    printf("write_file: open for path %s, fd=%d, errno=%d\n", path, fd, errno);
    if(fd == -1){
        fprintf(stderr, "write_file: open failed for path %s, err=%s, errno=%d\n", path, strerror(errno), errno);
        return 0;
    }
    auto bytesWritten = write(fd, data, dataSize);
    printf("bytes written to path %s=%ldl bytes\n", path, bytesWritten);
    close(fd);
    return bytesWritten;
}

}

int main(int argc, char *argv[]) {
    return 0;
}
`

lum1n0us, 2022-12-15:
Nice catch. It looks like an issue. We will track in #1814

wenyongh, 2023-04-24:
Fixed in #1816.

closed 2023-04-24 by wenyongh