wasm-micro-runtime/wasm-micro-runtime #1790: WASI threads support

loganek, 2022-12-07:
I create this issue to track implementation of the [wasi threads](https://github.com/WebAssembly/wasi-threads) proposal, define technical approach and give opportunity for discussion.

lum1n0us, 2022-12-12:
> Users who are willing to use threads, should be instructed to use wasi-libc implementation on top of wasi threads in WAMR, so eventually pthread library can be deprecated and removed.

Does it mean users have to write some glue code to satisfy *wasi-threads* requirements and can't keep their original version of code?

yamt, 2022-12-12:
> Does it mean users have to write some glue code to satisfy _wasi-threads_ requirements and can't keep their original version of code?

embedder will need wasi initialization etc.

wasm apps will need a rebuild at least.
it might or might not need some code changes to accommodate for incompatibilities between pthread implementations.

loganek, 2022-12-13:
> Does it mean users have to write some glue code to satisfy wasi-threads requirements and can't keep their original version of code?

Glue code might or might not be needed depending on the features being used. At the moment wasi-libc does not implement all the pthread API (having said that, I don't think WAMR pthread implement it all either?) so there might be some limitations.

I'd suggest keeping pthread as an option for now and only if we achieve feature parity in wasi-libc (or we're close to that, and we're ok accepting potential gaps), we can drop pthread library.

loganek, 2023-08-22:
I'm resolving the issue as the functionality has been already implemented and a lot of tests have been added. We'll continue testing and bugfixing the feature but any further problems will be tracked in separate github issue.

closed 2023-08-22 by loganek