wasm-micro-runtime/wasm-micro-runtime #1802: main args not passed to wasm buffer "main" function

blackbird1, 2022-12-11:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/14492757f87511eca405f386a4c71a624fb44f5d/core/iwasm/common/wasm_application.c#L110

blackbird1, 2022-12-11:
After looking at the spec it looks as though _start is supposed to call "main". The question is how can I pass args to my "main" function?

lum1n0us, 2022-12-14:
Please refer to samples, like:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/97d2b5a06082ad60f816c056d7aefb6f5ffcdc37/samples/basic/src/main.c#L173

wenyongh, 2022-12-15:
> After looking at the spec it looks as though _start is supposed to call "main". The question is how can I pass args to my "main" function?

Please use `wasm_runtime_set_wasi_args`, refer to:

https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/product-mini/platforms/posix/main.c#L616-L617

blackbird1, 2022-12-19:
So I'm using this public API:

https://github.com/bytecodealliance/wasm-micro-runtime/blob/14492757f87511eca405f386a4c71a624fb44f5d/core/iwasm/common/wasm_application.c#L202

I would expect "wasm_application_execute_main" should take the args I pass into it and pass them on to the "main" function.

wenyongh, 2022-12-26:
> So I'm using this public API:
> 
> https://github.com/bytecodealliance/wasm-micro-runtime/blob/14492757f87511eca405f386a4c71a624fb44f5d/core/iwasm/common/wasm_application.c#L202
> 
> I would expect "wasm_application_execute_main" should take the args I pass into it and pass them on to the "main" function.

Wondering why you want to manually call the main function with arguments? For wasi, it is supposed to call _start function:
_start calls __main_void
__main_void calls __wasi_args_sizes_get and __wasi_args_get to get arguments, and then call main

Maybe you can export main with `-Wl,--export=main` to have a try.

blackbird1, 2023-01-03:
I'm actually experimenting with using [openssl wasm](https://github.com/voltbuilder/openssl-wasm). Unfortunately this wasm buffer utilizes the main method just like the openssl command line. What is the recommended approach for this scenario?