wasm-micro-runtime/wasm-micro-runtime #1811: wasm_runtime_full_init API call can cause segmentation fault

cngzhnp, 2022-12-14:
**Compiler version:** GNU 10.2.1
**WASM version:** 1.1.1
**Problem Type:** Segmentation Fault

**Expected behaviour**
WASM API call should not cause any segmentation fault even though on edge cases.

**Actual behaviour**
`wasm_runtime_full_init` API call can cause segmentation fault when its parameter set to `nullptr`.

**Steps to reproduce the behaviour**
```
RuntimeInitArgs * args = nullptr;
if(!wasm_runtime_full_init(args))
{
   // Cannot reach here due to seg fault...
}
```
Cause segmentation fault which required to check whether parameter object has valid address or not before using its members.

yamt, 2023-01-05:
>  wasm_runtime_full_init API call can cause segmentation fault when its parameter set to nullptr.

this behavior seems perfectly fine.

cngzhnp, 2023-01-05:
IMHO, it is not. It should be explicitly declared in the API documentation, the API call parameter must be allocated before API call executed and its address must be valid and accessible. If this information was not provided, how user could know that who is responsible for the memory management? Maybe, it is allocated by itself inside the API call?

Also, API call should not cause any segmentation fault at least it should be guaranteed by "[design by contract](https://en.wikipedia.org/wiki/Design_by_contract)".