wasm-micro-runtime/wasm-micro-runtime #1831: Build should check for global libuv and uvwasi before downloading

phated, 2022-12-22:
I've run into some problems packaging WAMR for OCaml because libuv and uvwasi are **always** downloaded when setting `WAMR_BUILD_LIBC_UVWASI=1`. I believe that the CMake should be altered to locate libuv and uvwasi in the environment before downloading them. This is actually a feature of `FetchContent_MakeAvailable` after CMake 3.24 but I think WAMR allows an older version (and doesn't use `FetchContent_MakeAvailable`).

lum1n0us, 2022-12-23:
IIUC, 
- `cmake_minimum_required` just limited and check the minimum version of the cmake. It actually use the cmake on the host machine in the system path as long as it passed the requirement.
- `FetchContent_MakeAvailable`  equivalent logic can be obtained by calling  `FetchContent_GetProperties` and `FetchContent_Populate`.  But yes, 
>  it is simpler and provides additional features over the pattern above
- `FetchContent_MakeAvailable` is a new feature in CMake 3.14.

At first, I was also confused by the message *Fetching uvwasi ..* and *Featching libuv ..* and time pausing when cmake configuring. But after took a close look of the timestamp of a download file, `build/_deps/libuv-src/XXX` and `build/_deps/uvwasi-src/XXX`, I realized cmake actually reused downloaded files unless `build/_dpes/***` are removed.

phated, 2022-12-23:
You are correct that it will only be downloaded once, as long as sources aren't removed; however, some tools build in sandbox and clear all sources.

This issue is about using find_library before downloading anything, so libuv and uvwasi can be provided externally and are never downloaded.

lum1n0us, 2022-12-23:
True. it is better if we can store it externally. Let's see what we can do. thanks for asking.

phated, 2023-10-09:
This should be completed with the merge of #1861

closed 2023-10-09 by phated