wasm-micro-runtime/wasm-micro-runtime #1833: WASM C API (wasm_instance_new_with_args) fails with empty or null imports

blackbird1, 2022-12-23:
This assertion is cryptic and hard to understand. It fails with empty or null imports.

https://github.com/bytecodealliance/wasm-micro-runtime/blob/d04f3988c3e456788c72f44cf807b4af35d94968/core/iwasm/common/wasm_c_api.c#L4853

wenyongh, 2022-12-24:
Thanks for reporting the issue, seems that we had better remove the assertion.

lum1n0us, 2022-12-26:
@wenyongh  @blackbird1 actually, I am thinking about enhancing the assertion about `wasm_extern_vec_t imports` and preventing `wasm_instance_new` from invalid `imports`. 

from [the spec](https://webassembly.github.io/spec/core/exec/modules.html#instantiation):
> If the number  of imports is not equal to the number  of provided external values then: Fail.

wenyongh, 2022-12-26:
> @wenyongh @blackbird1 actually, I am thinking about enhancing the assertion about `wasm_extern_vec_t imports` and preventing `wasm_instance_new` from invalid `imports`.
>
> from [the [...]

Do you mean we change the assertion like:
```C
    if (import_func_count > 0)
        bh_assert((uint32)(func_import - *p_func_imports) == import_func_count); 
    }
```
Or
```C
    bh_assert(import_func_count == 0
              || (uint32)(func_import - *p_func_imports) == import_func_count); 
```
The above code won't be impacted by the these lines, right?
```C
        if (wasm_extern_kind(in) != WASM_EXTERN_FUNC)
            continue;
```
Since import_count is the total count of imports, and import_func_count is the total count of import functions. And if the latter is larger than 0, all the import functions must be linked, and `(uint32)(func_import - *p_func_imports) == import_func_count)` must be valid.

lum1n0us, 2022-12-26:
- add a checker at [the beginning](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/common/wasm_c_api.c#L4730). If `imports` doesn't match the module requirement, like length, fail.
- https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/common/wasm_c_api.c#L4853 should be lik `bh_assert(import_func_count == 0 || (uint32)(func_import - *p_func_imports) == import_func_count); `

lum1n0us, 2022-12-28:
#1843

closed 2023-04-23 by wenyongh