wasm-micro-runtime/wasm-micro-runtime #1835: parse names failed

qianyimao, 2022-12-24:
mismatched index when parse the names which build by AssebmlyScript

core/iwasm/interpreter/wasm_loader.c ! handle_name_section
--- read_leb_uint32(p, p_end, func_index);
+++read_leb_uint32(p, p_end, func_index);
+++func_index += module->import_global_count;

wenyongh, 2022-12-26:
@qianyimao A little strange why add `module->import_global_count` into `func_index`? Could you upload the wasm binary file or give more details about how to reproduce the issue?

qianyimao, 2022-12-26:
"some1.value" causes "WASMModule::import_global_count" to 1
for loading the wasm file must resolve "some1.value" in engine.

script of AssemblyScript:
@external("some1", "value")
declare let value: i32;

export function main(): void
{
    console.log(`${value}`)
}

dumped by wasm-objdump.exe
Import[3]:
 - func[0] sig=6 <~lib/builtins/abort> <- env.abort
 - func[1] sig=2 <~lib/bindings/dom/console.log> <- env.console.log
 - global[0] i32 mutable=1 <- some1.value
Function[63]:

wenyongh, 2022-12-27:
Hi, I tested the wasm app as you mentioned, but didn't find any issue with the func index and func name resolved:
The patch to dump the func index and func name in wasm loader handle_name_section:
```diff
--- a/core/iwasm/interpreter/wasm_loader.c
+++ b/core/iwasm/interpreter/wasm_loader.c
@@ -2791,6 +2791,7 @@ handle_name_section(const uint8 *buf, const uint8 *buf_end, WASMModule *module,
                     for (name_index = 0; name_index < num_func_name;
                          name_index++) {
                         read_leb_uint32(p, p_end, func_index);
+                        printf("##func_index: %u\n", func_index);
                         if (func_index == previous_func_index) {
                             set_error_buf(error_buf, error_buf_size,
                                           "duplicate function name");
@@ -2820,6 +2821,7 @@ handle_name_section(const uint8 *buf, const uint8 *buf_end, WASMModule *module,
                                           error_buf_size))) {
                                 return false;
                             }
+                            printf("##func name: %s\n", module->functions[func_index]->field_name);
                         }
                         p += func_name_len;
                     }
```
And build iwasm with `cmake .. -DWAMR_BUILD_CUSTOM_NAME_SECTION=1`.

The asc source code of wasm app:
```asc
import "wasi"
@external("some", "value")
declare let value: i32;

export function main(): void
{
    console.log(value.toString())
}

main()
```
And build it with `asc --debug main.ts -o test.wasm`

The result of `iwasm test.wasm`:
```bash
##func_index: 0
##func_index: 1
##func_index: 2
##func_index: 3
##func name: ~lib/string/String#get:length
##func_index: 4
##func name: ~lib/string/String.UTF8.encodeUnsafe
##func_index: 5
##func name: ~lib/string/String.UTF8.encodeUnsafe@varargs
##func_index: 6
##func name: ~lib/util/number/decimalCount32
##func_index: 7
##func name: ~lib/bindings/wasi_snapshot_preview1/iovec#set:buf_len
##func_index: 8
##func name: ~lib/wasi/index/abort
...
```
Note that func 0 to 2 are the import functions, their name isn't resolved by handle_name_section.

And the result of `wasm-objdump -x test.wasm`:
```
Custom:
 - name: "name"
 - func[0] <~lib/bindings/wasi_snapshot_preview1/fd_write>
 - func[1] <~lib/bindings/wasi_snapshot_preview1/proc_exit>
 - func[2] <~lib/bindings/wasi_snapshot_preview1/iovec#set:buf>
 - func[3] <~lib/string/String#get:length>
 - func[4] <~lib/string/String.UTF8.encodeUnsafe>
 - func[5] <~lib/string/String.UTF8.encodeUnsafe@varargs>
 - func[6] <~lib/util/number/decimalCount32>
 - func[7] <~lib/bindings/wasi_snapshot_preview1/iovec#set:buf_len>
 - func[8] <~lib/wasi/index/abort>
 ...
 ```
From the result we can see that the name of func 3 to func 8 (and more) dumped by iwasm and wasm-objdump are the same.

qianyimao, 2022-12-28:
put a breakpoint on "wasm_interp_classic.c, line 3871", watch the updating of "cur_func.u.func.field_name" on each hitted. you will find that the cur_func had a wrong field_name. fixing the field_name by adding the "func_index += module->import_global_count".
version is WAMR-1.1.2.

qianyimao, 2022-12-28:
#define WASM_ENABLE_INTERP                  1
#define WASM_ENABLE_AOT                     1
#define WASM_ENABLE_FAST_INTERP             0
#define WASM_ENABLE_BULK_MEMORY             1
#define WASM_ENABLE_DUMP_CALL_STACK         0
#define WASM_ENABLE_LOAD_CUSTOM_SECTION     1
#define WASM_ENABLE_MULTI_MODULE            1
#define WASM_ENABLE_SHARED_MEMORY           0
#define WASM_ENABLE_MINI_LOADER             0
#define WASM_ENABLE_CUSTOM_NAME_SECTION     1
#define WASM_DISABLE_HW_BOUND_CHECK         1
#define WASM_DISABLE_STACK_HW_BOUND_CHECK   1
#define WASM_ENABLE_LIBC_BUILTIN            1
#define WASM_ENABLE_LIBC_WASI               0
#define WASM_ENABLE_UVWASI                  0
#define WASM_ENABLE_DEBUG_INTERP            1
#define WASM_ENABLE_THREAD_MGR              1

wenyongh, 2022-12-30:
@qianyimao Thanks a lot for reporting the issue! I submitted PR #1849 to fix, please check whether it works for you.

closed 2022-12-30 by qianyimao