wasm-micro-runtime/wasm-micro-runtime #1860: wasm_cluster_thread_waiting_run maybe blocked

qianyimao, 2023-01-04:
wasm_cluster_thread_waiting_run will miss notification, if "wasm_cluster_thread_step" or "wasm_cluster_thread_continue" been invoked before "os_cond_wait" in wasm_cluster_thread_waiting_run. the pthread_cond_t is not signal safe and "stop event" can been finished as quick as possible by gdbserver.

qianyimao, 2023-01-04:
@wenyongh

xujuntwt95329, 2023-01-06:
Hi @qianyimao, thanks for reporting this issue.

Yes, by design, the `interpreter thread` and the `debugger thread` should not execute at the same time, so we don't do too much synchronization between them. The issue is because there is a very short time they do execute in parallel, which may cause the signal to be missed.

I'll fix the issue today or tomorrow. Thanks again!

wenyongh, 2023-01-16:
@qianyimao We have submitted PR #1887 to fix this issue, could you help review and test it?

closed 2023-01-18 by qianyimao