wasm-micro-runtime/wasm-micro-runtime #1899: Missing fast-jit atomic operations

eloparco, 2023-01-18:
`cmpxchg`, `wait` and `notify` instructions are currently not supported in fast-jit mode, see https://github.com/bytecodealliance/wasm-micro-runtime/blob/cb7ac8fee3473418bc717246051636fb59475e2d/core/iwasm/fast-jit/fe/jit_emit_memory.c#L775
Those operations are used in libc
https://github.com/WebAssembly/wasi-libc/blob/a00bf321eeeca836ee2a0d2d25aeb8524107b8cc/libc-bottom-half/crt/crt1-command.c#L22 and therefore currelny necessary to use WASI threads with fast-jit https://github.com/bytecodealliance/wasm-micro-runtime/blob/5f0f8fe7ee04bbc9e66df2a604a06b241f2fcc97/.github/workflows/compilation_on_android_ubuntu.yml#L326

TianlongLiang, 2023-01-18:
Hi, we are planning to add support for Threads opcode in fast-jit mode. I will open a new issue to track the development process by then. I will also keep you updated if you are interested in it 😄

eloparco, 2023-01-18:
I started taking a look at it too, I'll create a PR if I get it to work.

eloparco, 2023-01-19:
I managed to get a basic implementation of `cmpxchg` (only the `i32.atomic.rmw.cmpxchg` for now), but I'm struggling with the `memory.atomic.wait32`.
I'm trying reproduce what we already have in https://github.com/bytecodealliance/wasm-micro-runtime/blob/cb7ac8fee3473418bc717246051636fb59475e2d/core/iwasm/interpreter/wasm_interp_classic.c#L3403
and that's what I got so far:
```c
bool
jit_compile_op_atomic_wait(JitCompContext *cc, uint8 op_type, uint32 align,
                           uint32 offset, uint32 bytes)
{
    assert(op_type == VALUE_TYPE_I32 && "wait64 not implemented yet");

    JitReg timeout, expect, addr, res, memory_data, offset1, expect_64,
        addr_ptr;
    JitReg args[5] = { 0 };

    POP_I64(timeout);
    POP_I32(expect);
    POP_I32(addr);

    expect_64 = jit_cc_new_reg_I64(cc);
    GEN_INSN(I32TOI64, expect_64, expect);

    offset1 = check_and_seek(cc, addr, offset, bytes);
    if (!offset1) {
        goto fail;
    }
    memory_data = get_memory_data_reg(cc->jit_frame, 0);
    addr_ptr = jit_cc_new_reg_ptr(cc);
    GEN_INSN(LDPTR, addr_ptr, memory_data, offset1);

    res = jit_cc_new_reg_I32(cc);
    args[0] = get_module_inst_reg(cc->jit_frame);
    args[1] = addr_ptr;
    args[2] = expect_64;
    args[3] = timeout;
    args[4] = NEW_CONST(I32, false);

    if (!jit_emit_callnative(cc, wasm_runtime_atomic_wait, res, args,
                             sizeof(args) / sizeof(args[0])))
        goto fail;

    GEN_INSN(CMP, cc->cmp_reg, res, NEW_CONST(I32, -1));
    if (!jit_emit_exception(cc, EXCE_ALREADY_THROWN, JIT_OP_BEQ, cc->cmp_reg,
                            NULL))
        goto fail;

    PUSH_I32(res);

    return true;
fail:
    return false;
}
```

Unfortunately, I see [`wasm_runtime_atomic_wait`](https://github.com/bytecodealliance/wasm-micro-runtime/blob/cb7ac8fee3473418bc717246051636fb59475e2d/core/iwasm/common/wasm_shared_memory.c#L314) being called with `address` equal to `null`, while the other parameters seem ok.

Any hints on what I might be doing wrong? Thanks

eloparco, 2023-01-19:
Actually, I just found out that if I instead use `args[1] = memory_data;` I get an `address` different from `null`. Now I need to understand how to add the offset.

eloparco, 2023-03-26:
Started in #1903 and continued in #1984.

closed 2023-03-26 by eloparco