wasm-micro-runtime/wasm-micro-runtime #1906: Is non-SGX still sandboxed?

rygo6, 2023-01-22:
I notice the main page says `TEE(Trusted Execution Environment) support(SGX/TDX)`
https://bytecodealliance.github.io/wamr.dev/

Does that mean WAMR can only run untrusted code on Intel processors through its SGX support? Or is WAMR still appropriately sandboxed with SGX disabled?

wenyongh, 2023-01-23:
Hi, WAMR supports many platforms, such as linux, linux-sgx, macos, windows, android and so on, you can build and run WAMR without SGX, refer to: 
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#linux
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wamr.md#macos

xwang98, 2023-01-31:
Without SGX, WAMR is still sandbox. @rygo6

closed 2023-04-19 by wenyongh