wasm-micro-runtime/wasm-micro-runtime #1935: Building the runtime with an out-of-repository platform is hacky

cookpate, 2023-02-03:
I want to build WAMR's runtime with custom, internal platforms, but our team does not want to maintain a patch set or fork of the repository.

Here's how `runtime_lib.cmake` uses some key CMake variables in the current release (1.1.2):
```cmake
include (${SHARED_DIR}/platform/${WAMR_BUILD_PLATFORM}/shared_platform.cmake)
include (${SHARED_DIR}/mem-alloc/mem_alloc.cmake)
include (${IWASM_DIR}/common/iwasm_common.cmake)
include (${SHARED_DIR}/utils/shared_utils.cmake)
```

I could touch `SHARED_DIR`, but this would require us to also maintain `mem_alloc.cmake` and `shared_utils.cmake` in our own repository.

I could touch `WAMR_BUILD_PLATFORM` to use a relative path. For example: 
```cmake
set (WAMR_BUILD_PLATFORM ../../../../MyRepo/WAMR)
```
But, this is a hack and breaks if the directory structure changes in depth anywhere. It also breaks and string parsing on `WAMR_BUILD_PLATFORM` for configuration.

I talked with the team, and one solution we propose is setting another variable to set the absolute directory path to `shared_platform.cmake` independent of both `WAMR_BUILD_PLATFORM` and `SHARED_DIR`.

loganek, 2023-02-03:
I think it makes sense to enable that.

On aside note, can you share what's the platform? Would you consider upstreaming the implementation?

cookpate, 2023-02-04:
Our platform is made for internal use and is built on top of non-public APIs.

PR #1941 (2023-02-07): Enable specifying out-of-source platform configuration cmake file
resolves #1935
commit 251e63f3 (2023-02-07): Enable specifying out-of-source platform configuration cmake file (#1941)

Resolve #1935, enable specifying out-of-source platform folder with
`cmake .. -DWAMR_BUILD_PLATFORM=new-os
               -DSHARED_PLATFORM_CONFIG=/path/to/new-os/shared_platform.cmake`
closed 2023-02-07 by wenyongh