wasm-micro-runtime/wasm-micro-runtime #1953: Release binary iwasm does not have execution modes

matsbror, 2023-02-12:
According to this blog post: https://bytecodealliance.github.io/wamr.dev/blog/introduction-to-wamr-running-modes/ there should be different execution modes in the binary release, but the downloaded binary does not have these modes:

```
iwasm.exe --fast-jit generichash.wasm
Usage: iwasm [-options] wasm_file [args...]
options:
  -f|--function name       Specify a function name of the module to run rather
                           than main
  -v=n                     Set log verbose level (0 to 5, default is 2) larger
                           level with more log
  --stack-size=n           Set maximum stack size in bytes, default is 64 KB
  --heap-size=n            Set maximum heap size in bytes, default is 16 KB
  --repl                   Start a very simple REPL (read-eval-print-loop) mode
                           that runs commands in the form of "FUNC ARG..."
  --env=<env>              Pass wasi environment variables with "key=value"
                           to the program, for example:
                             --env="key1=value1" --env="key2=value2"
  --dir=<dir>              Grant wasi access to the given host directories
                           to the program, for example:
                             --dir=<dir1> --dir=<dir2>
  --addr-pool=<addrs>      Grant wasi access to the given network addresses in
                           CIRD notation to the program, seperated with ',',
                           for example:
                             --addr-pool=1.2.3.4/15,2.3.4.5/16
  --allow-resolve=<domain> Allow the lookup of the specific domain name or domain
                           name suffixes using a wildcard, for example:
                           --allow-resolve=example.com # allow the lookup of the specific domain
                           --allow-resolve=*.example.com # allow the lookup of all subdomains
                           --allow-resolve=* # allow any lookup
  --native-lib=<lib>       Register native libraries to the WASM module, which
                           are shared object (.so) files, for example:
                             --native-lib=test1.so --native-lib=test2.so
  --max-threads=n          Set maximum thread number per cluster, default is 4
  --version                Show version information
```

TianlongLiang, 2023-02-13:
Hi, we planned to compile this feature into `iwasm` in the next release. Sorry to confuse you in the blog, we will update it and make it clearer right away.

If you would like to try it now, kindly use the following commands to compile an equivalent `iwasm` shown in our blog:

```sh
# in directory {WAMR Directory}/product-mini/platforms/{your platform}
mkdir -p build
cmake -DWAMR_BUILD_FAST_JIT=1 -DWAMR_BUILD_JIT=1 -B build
cmake --build build
```

`iwasm` will be in the build folder.

wenyongh, 2023-04-07:
@matsbror WAMR-1.2.1 was released with running mode control supported: https://github.com/bytecodealliance/wasm-micro-runtime/releases/tag/WAMR-1.2.1

closed 2023-04-23 by wenyongh