wasm-micro-runtime/wasm-micro-runtime #1979: Unexpected `proc_exit` behaviour in AOT mode

csegarragonz, 2023-02-21:
Hi,

I am cross-compiling the following C program:

```c
// exit.c
#include <stdlib.h>

int main(int argc, char** argv)
{
    exit(0);
}
```

I use  WASI-SDK version 19 to cross-compile it to WebAssembly:

```bash
/opt/wasi-sdk/bin/clang --sysroot=/opt/wasi-sdk/share/wasi-sysroot exit.c -o exit.wasm
```

I then build `wamrc` and `iwasm` with the default settings on a Linux x86_64 machine from commit [37b09d](https://github.com/bytecodealliance/wasm-micro-runtime/commits/37b09d0f240c411e571dda39eb0ad231d6068637).

When running in interpreter mode, `iwasm` finishes correctly and returns a 0 exit code:

```bash
iwasm exit.wasm
echo $?
# 0
```

However, when running in AOT mode:

```bash
wamrc -o exit.aot exit.wasm
iwasm exit.aot
# Exception: unknown exception
echo $?
# 1
```

Do you know what could be going on?

TianlongLiang, 2023-03-02:
Hi, I tried your attached file on the latest commit, and it exits as expected. Can you try it on the latest commit, too, to see whether this behavior persists?

csegarragonz, 2023-03-02:
It does work now on the latest commit!

Thank you very much for the follow up :))

closed 2023-03-02 by csegarragonz