wasm-micro-runtime/wasm-micro-runtime #1982: Issues building wamr compiler on ARM64

matsbror, 2023-02-22:
I am trying to build iwasm and wamr compiler on an ARM64 Linux machine on AWS and have issues building `wamrc` depending on if I use clang or gcc. I could build `iwasm` without any issues. However I have not yet been able to build the AOT compielr. 

With gcc (11.3) I get:

```
% CC=gcc CXX=g++ cmake ..
% make
gcc: error: unrecognized command-line option ‘-mindirect-branch-register’
```

With clang (14) as compiler I get:

```
% CC=clang CXX=clang++ cmake ..
% make
...
[  3%] Building C object CMakeFiles/vmlib.dir/home/ubuntu/src/wasm-micro-runtime/core/shared/platform/common/posix/posix_memmap.c.o
/home/ubuntu/src/wasm-micro-runtime/core/shared/platform/common/posix/posix_memmap.c:74:22: error: use of undeclared identifier 'MAP_32BIT'
        map_flags |= MAP_32BIT;
```

Any pointers?

matsbror, 2023-02-23:
# UPDATE 2

I was able to build the aot compiler by setting `-DWAMR_BUILD_TARGET=AARC64` to cmake argument.

However, then I have the following issue when trying to run the aot-compiled program:
```
ubuntu@ip-172-31-40-38:~/wabench/JetStream2/gcc-loops$ iwasm gcc-loops.aot
AOT module load failed: invalid target type, expected aarch64.v8 but got aarch64
```

I made sure both `iwasm` and `wamrc` was built with the same build target and I tried also changing build target to aarch64.v8 on both with the same result. I have attached the wasm and the aot files here for your perusal.

[gcc-loops.tar.gz](https://github.com/bytecodealliance/wasm-micro-runtime/files/10811363/gcc-loops.tar.gz)

matsbror, 2023-02-23:
# UPDATE 3

I discovered the `--target` option of `wamrc` but it gave another dimension to the problem:

```
ubuntu@ip-172-31-40-38:~/wabench/JetStream2/gcc-loops$ wamrc --target=aarch64v8 -o gcc-loops.aot gcc-loops.wasm
Create AoT compiler with:
  target:        aarch64v8
  target cpu:
  cpu features:
  opt level:     3
  size level:    3
  output format: AoT file
Compile success, file gcc-loops.aot was generated.
%:~/wabench/JetStream2/gcc-loops$ iwasm gcc-loops.aot
AOT module load failed: invalid target type, expected aarch64.v8 but got aarch64v8
%:~/wabench/JetStream2/gcc-loops$ wamrc --target=aarch64.v8 -o gcc-loops.aot gcc-loops.wasm
Error: Invalid target. Use --target=help to list all supported targets
```

closed 2023-02-23 by matsbror

JamesMenetrey, 2023-02-27:
Thanks for sharing the solution @matsbror!