wasm-micro-runtime/wasm-micro-runtime #1990: lib_export.h not installed?

behdad, 2023-02-25:
On linux when I do `cmake install`, I see the following files installed:
```
-- Install configuration: "Release"
-- Up-to-date: /usr/local/lib/libvmlib.a
-- Up-to-date: /usr/local/lib/libiwasm.so
-- Up-to-date: /usr/local/include/wasm_c_api.h
-- Up-to-date: /usr/local/include/wasm_export.h
```
However, `wasm_export.h` includes `lib_export.h`. So I had to manually install that file. Am I missing something?

wenyongh, 2023-02-28:
Yes, thanks for reporting the issue, we should add `lib_export.h` in the copy file list of CMakeLists.txt:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/CMakeLists.txt#L161-L162

wenyongh, 2023-03-03:
Fixed in #1996

closed 2023-03-03 by behdad