wasm-micro-runtime/wasm-micro-runtime #2005: aot floating data not working on thumbv7

wufan0920, 2023-03-06:
I encountered similar issue as [aot not working on armv7a](https://github.com/bytecodealliance/wasm-micro-runtime/issues/1582) But my problem happens on thumbv7(mimxrt1060evk).
I tried to follow the [workaround](https://github.com/bytecodealliance/wasm-micro-runtime/issues/1582#issuecomment-1277422820) and [another workaround](https://github.com/bytecodealliance/wasm-micro-runtime/issues/1582#issuecomment-1277569537) from the issue mentioned above to handle this but failed.
any idea for the issue?

BeiMingYouPeng, 2023-03-09:
i have the same issue to run the samples/basic demo in the AOT mode on the i.mxRT1060 EVK board. the generate_float() wasm function failed.

BeiMingYouPeng, 2023-03-09:
tested with these params, but not work
wamrc --target=thumbv7 --target-abi=eabi --cpu=cortex-m7 -o testapp.aot testapp.wasm
 wamrc --target=thumbv7em --target-abi=gnu --cpu=cortex-m7 -o testapp.aot testapp.wasm

FromLiQg, 2023-03-10:
Hi ~
Can you share the detailed error information? I tried the aot of Thumbv7 on Cortex-A53 and found no problem.

BeiMingYouPeng, 2023-03-10:
> Hi ~ Can you share the detailed error information? I tried the aot of Thumbv7 on Cortex-A53 and found no problem.

yes,  I tested the same samples/basic demo on RaspberryPi 4B. No issue found.  and it also works on x86_64.   
But it is with issue on the i.mxRT1060 EVK board (NXP arm cortex m7).  the interpreter mode works fine on the evk board, but with aot mode, then it fails.
this the correct output when in interpreter mode:
[image]

but in aot mode: it stuck here
[image]
failed to run wasm function "generate_float"

below is the log run on raspberryPi, the aot mode is ok:
[image]

FromLiQg, 2023-03-10:
Hi~
I also tested the program on the raspberryPi 3B+, and found that there was no problem.
Since I don't have the development board of Cortex-m7, I may not be able to test the program myself.
I recommend the following ideas:
1. Check whether the functions are consistent
* functions in `invokeNative_xxx.s`
* `wasm_runtime_invoke_native`

shell instructions:
```shell
wamrc --format=object -o test.o testapp.wasm
objdump -d test.o > test.log
```
2. Single-step debugging to see where the program is blocked.
3. add this option to build `aot` and `iwasm`
```shell
# to build `aot`
wamrc --target=THUMBV7_VFP

# to build iwasm
cmake .. -DWAMR_BUILD_TARGET=THUMBV7_VFP
```

BeiMingYouPeng, 2023-03-12:
@FromLiQg 
>1.  Check whether the functions are consistent

- via debug i can see it calls the iwasm/common/arch/invokeNative_thumb.s for arguments pass,
- but via the objdump, i do not get your point how to check they are consistent. 
i attached the logs here
[objudump-of-testapp.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/10949984/objudump-of-testapp.txt)
[debugLog.txt](https://github.com/bytecodealliance/wasm-micro-runtime/files/10949985/debugLog.txt)

>2 Single-step debugging to see where the program is blocked.

-  please also see the debug log which attached above. it blocked at invokeNative_thumb.s:83 during call of the wasm function
- the native invoke was step in from the line i marked red in below figure
[image]

>3 add this option to build aot and iwasm

- the VFP is not supported with below error during compile
`core/iwasm/common/arch/invokeNative_thumb_vfp.s:57: Error: Instruction not permitted on this architecture`

and more information that, if i changed the wasm function from float into int, then it works fine
```
#if 0
float
generate_float(int iteration, double seed1, float seed2)
{
    float ret;
    printf("calling into WASM function: %s\n", __FUNCTION__);
    for (int i = 0; i < iteration; i++) {
        ret += 1.0f / seed1 + seed2;
    }
    return ret;
}
#endif
int
generate_float(int iteration, int seed1, int seed2)
{
    int ret;
    printf("calling into WASM function: %s\n", __FUNCTION__);
    ret = iteration + seed1 + seed2;
    return ret;
}
```

FromLiQg, 2023-03-14:
@BeiMingYouPeng 

Hi~ thank you for your log.
As you said, it blocked at `blx ip`.
So I spent some days investigating the instruction `blx`.
[ARMv7-M Architecture Reference Manual](https://documentation-service.arm.com/static/606dc36485368c4c2b1bf62f)

I guess it may be that the lowest bit of the function address is 0 causes this error. If the lowest bit of the function address is 0, it will cause the instruction set to switch from Thumb to ARM, which will cause an **exception** in Cortex-M7.(see the file in page `A7-214` introducing `blx`, please )

The lowest bit of the value of the register is used to update the `T` bit of the `CPSR` register when `blx` execute. If it is 1, it enters the `Thumb` state. If it is 0, enter the `ARM` state.

Can you debug it again, and print the value of `ip` register, and see if the lowest point is 0?
In addition, I have no more clues for the time being.

BeiMingYouPeng, 2023-03-15:
@FromLiQg 
thanks for the support. 
I figured out the root cause of this issue. it is not in WAMR, but it is the configuration of my Zephyr.
i attached the information below in case others face similar problem 
1. in the zephyr default configuration for the nxp i.mxRT1060 evk board, the FPU is not enabled. enable it 
```
in the boards/arm/mimxrt1060_evk/mimxrt1060_evk_defconfig add below two lines

CONFIG_FPU=y
CONFIG_FPU_SHARING=y
```
2. for the application compilation, add the ` --target-abi=gnueabihf` for floating point
3. for the wamr runtime build,  add the ` -DWAMR_BUILD_TARGET=THUMBV7EM_VFP`， 
note, in my previous comment, which i mentioned the "Error: Instruction not permitted on this architecture",  it is due to the FPU is not enabled. once FPU enabled, then it is without this error when build with "_VFP"

thanks again

FromLiQg, 2023-03-15:
@BeiMingYouPeng 
Thank you for your feedback. I'm glad to see that you have solved the problem

closed 2023-03-15 by wufan0920