wasm-micro-runtime/wasm-micro-runtime #2007: Confusing comments in invokeNative_aarch64.s

ApsarasX, 2023-03-06:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/e6a01847971747824e7ed6dd27d4d3b6bdbf9ece/core/iwasm/common/arch/invokeNative_aarch64.s#L34-L44

All float-pointing here occupies 16 bytes. So I think the above code should be
```diff
        /* Fill in float-point registers */
-        ldp     d0, d1, [x20], #16 /* d0 = argv[0], d1 = argv[1] */
+        ldp     d0, d1, [x20], #16 /* d0 = argv[0], d1 = argv[2] */
-        ldp     d2, d3, [x20], #16 /* d2 = argv[2], d3 = argv[3] */
+        ldp     d2, d3, [x20], #16 /* d2 = argv[4], d3 = argv[6] */
-        ldp     d4, d5, [x20], #16 /* d4 = argv[4], d5 = argv[5] */
+        ldp     d4, d5, [x20], #16 /* d4 = argv[8], d5 = argv[10] */
-        ldp     d6, d7, [x20], #16 /* d6 = argv[6], d7 = argv[7] */
+        ldp     d6, d7, [x20], #16 /* d6 = argv[12], d7 = argv[14] */

        /* Fill integer registers */
-        ldp     x0, x1, [x20], #16 /* x0 = argv[8] = exec_env, x1 = argv[9] */
+        ldp     x0, x1, [x20], #16 /* x0 = argv[16] = exec_env, x1 = argv[17] */
-        ldp     x2, x3, [x20], #16 /* x2 = argv[10], x3 = argv[11] */
+        ldp     x2, x3, [x20], #16 /* x2 = argv[18], x3 = argv[19] */
-        ldp     x4, x5, [x20], #16 /* x4 = argv[12], x5 = argv[13] */
+        ldp     x4, x5, [x20], #16 /* x4 = argv[20], x5 = argv[21] */
-        ldp     x6, x7, [x20], #16 /* x6 = argv[14], x7 = argv[15] */
+        ldp     x6, x7, [x20], #16 /* x6 = argv[22], x7 = argv[23] */
```

FromLiQg, 2023-03-06:
Hi~ 
These registers used here are 32-bit.

FromLiQg, 2023-03-06:
Hi~ Let me try to explain this code.
The `argv` here is not `uint32` type, but `uint64`.
For understanding this point ,we need to see this:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/common/wasm_runtime_common.c#L4156

zoowii, 2023-03-06:
@FromLiQg 
See this line: https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/common/wasm_runtime_common.c#L4202

```
    fps = (v128 *)argv1;
    ints = (uint64 *)(fps + MAX_REG_FLOATS);
```
fps is v128*, and MAX_REG_FLOATS is 8, so ints - argv1 = 8*2=16

FromLiQg, 2023-03-06:
See this line:
We found that it enables SIMD.
Then we need to use this `.s` file: invokeNative_aarch64_simd.s, but not invokeNative_aarch64.s

https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/common/arch/invokeNative_aarch64_simd.s

FromLiQg, 2023-03-21:
Hi~ @zoowii @ApsarasX 
Did this answer help you?

closed 2023-04-19 by wenyongh