wasm-micro-runtime/wasm-micro-runtime #2033: `load_module` hangs forever for JIT module

pbeza, 2023-03-15:
My application hangs forever while loading a JIT module. This problem appeared very recently (1-2 days ago). I'm using the `master` branch of WAMR. I have a feeling it might be related to the one of these 2 commits:
- https://github.com/bytecodealliance/wasm-micro-runtime/commit/578fbc5a55276576f3cc68742e31275e95535c2b
- https://github.com/bytecodealliance/wasm-micro-runtime/commit/bab2402b6e9d4990094d3226b3fb39f83528aa78
Notice that before https://github.com/bytecodealliance/wasm-micro-runtime/commit/578fbc5a55276576f3cc68742e31275e95535c2b version of the `asmjit` was not pinned to the specific commit. I think that fact may be relevant in this context.

I'm not able to share a full minimal code sample that fails, but this is a snippet of the code that hangs infinitely:
```
    std::cerr << "test_06\n";

    if(!(wasm_file_buf = (uint8_t*)read_file_to_buffer(wasm_file_path.c_str(), &wasm_file_size))) {
        return NULL;
    }

    std::cerr << "test_07\n";

    void* wasm_module;
    if(!(wasm_module = load_module(wasm_file_buf, wasm_file_size, error_buf, sizeof(error_buf)))) {  // HERE it hangs forever here
        printf("%s\n", error_buf);
        return NULL;
    }

    std::cerr << "test_08\n";
```
`load_module` is the function from [`product-mini/platforms/linux-sgx/enclave-sample/App/App.cpp`](https://github.com/bytecodealliance/wasm-micro-runtime/blob/26728cbef24ddf6c5d013904463192d4ec7009ec/product-mini/platforms/linux-sgx/enclave-sample/App/App.cpp#L401)

This is relevant `strace` log for the above code snippet:
```
openat(AT_FDCWD, "my_JIT_file.wasm", O_RDONLY) = 3
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=15112158, ...}, AT_EMPTY_PATH) = 0
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=15112158, ...}, AT_EMPTY_PATH) = 0
lseek(3, 15110144, SEEK_SET)            = 15110144
read(3, "nth_to_secs\363\35\f__secs_to_tm\364\35\16__y"..., 2014) = 2014
lseek(3, 0, SEEK_SET)                   = 0
mmap(NULL, 15114240, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f6a08953000
read(3, "\0asm\1\0\0\0\1\212\5M`\1\177\0`\1\177\1\177`\2\177\177\1\177`\3\177\177\177"..., 15110144) = 15110144
read(3, "nth_to_secs\363\35\f__secs_to_tm\364\35\16__y"..., 4096) = 2014
close(3)                                = 0
write(2, "test_07\n", 8test_07
)                = 8
rt_sigaction(SIGRT_1, {sa_handler=0x7f6a0a5f58f0, sa_mask=[], sa_flags=SA_RESTORER|SA_ONSTACK|SA_RESTART|SA_SIGINFO, sa_restorer=0x7f6a0a5a6520}, NULL, 8) = 0
rt_sigprocmask(SIG_UNBLOCK, [RTMIN RT_1], NULL, 8) = 0
mmap(NULL, 8392704, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS|MAP_STACK, -1, 0) = 0x7f6a08152000
mprotect(0x7f6a08153000, 8388608, PROT_READ|PROT_WRITE) = 0
rt_sigprocmask(SIG_BLOCK, ~[], [], 8)   = 0
clone3({flags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID, child_tid=0x7f6a08952910, parent_tid=0x7f6a08952910, exit_signal=0, stack=0x7f6a08152000, stack_size=0x7fff00, tls=0x7f6a08952640}, 88) = -1 ENOSYS (Function not implemented)
clone(child_stack=0x7f6a08951ef0, flags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID, parent_tid=[11417], tls=0x7f6a08952640, child_tidptr=0x7f6a08952910) = 11417
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
mmap(NULL, 8392704, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS|MAP_STACK, -1, 0) = 0x7f6a037ff000
mprotect(0x7f6a03800000, 8388608, PROT_READ|PROT_WRITE) = 0
rt_sigprocmask(SIG_BLOCK, ~[], [], 8)   = 0
clone3({flags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID, child_tid=0x7f6a03fff910, parent_tid=0x7f6a03fff910, exit_signal=0, stack=0x7f6a037ff000, stack_size=0x7fff00, tls=0x7f6a03fff640}, 88) = -1 ENOSYS (Function not implemented)
clone(child_stack=0x7f6a03ffeef0, flags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID, parent_tid=[11418], tls=0x7f6a03fff640, child_tidptr=0x7f6a03fff910) = 11418
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
futex(0x7f6a04000b90, FUTEX_WAIT, 4294967295, {tv_sec=5, tv_nsec=0}) = 0
```

It hangs infinitely on the last line, i.e. on:
```
futex(0x7f6a04000b90, FUTEX_WAIT, 4294967295, {tv_sec=5, tv_nsec=0}) = 0
```
It seems for me that some thread got `SIGRT_1` signal and terminated (sounds like something related to this commit https://github.com/bytecodealliance/wasm-micro-runtime/commit/bab2402b6e9d4990094d3226b3fb39f83528aa78 that addresses thread mgr issue).

Those are the flags I'm using to build WAMR:
```
ARG WAMR_BUILD_FLAGS="-DWAMR_BUILD_DUMP_CALL_STACK=1 \
  -DWAMR_BUILD_CUSTOM_NAME_SECTION=1 \
  -DWAMR_BUILD_BULK_MEMORY=1 \
  -DWAMR_BUILD_MULTI_MODULE=1 \
  -DWAMR_BUILD_SGX_IPFS=0 \
  -DWAMR_BUILD_LIB_RATS=1 \
  -DWAMR_BUILD_REF_TYPES=1 \
  -DWAMR_BUILD_FAST_JIT=1"
```
(copied-pasted from my `Dockerfile`)

Do you think it can be a bug? Do you have some tests for Fast JIT that should be able to detect this kind of bugs early?

I would appreciate any comments on this. Thanks.

TianlongLiang, 2023-03-16:
Hi, thanks for bringing this issue to our attention. I tried your compilation option without -DWAMR_BUILD_LIB_RATS=1 on the latest commit on the main branch and it didn’t hang. I am installing the SGX SSL library now and will try your complete compilation flags later. Meanwhile, can you try to compile iwasm without -DWAMR_BUILD_LIB_RATS=1 and see if it still hang on the latest commit?

TianlongLiang, 2023-03-16:
And can you provide your docker file if there is not too much sensitive information in it?

pbeza, 2023-03-16:
> Meanwhile, can you try to compile iwasm without -DWAMR_BUILD_LIB_RATS=1 and see if it still hang on the latest commit?

Unfortunately I'm using `librats` in my application and it's not trivial build it without `-DWAMR_BUILD_LIB_RATS=1`.

> And can you provide your docker file if there is not too much sensitive information in it?

Unfortunately my Dockerfile is based on some other proprietary image that I'm not able to share. I would need to spend some time on building a minimal example that would prove the problem.

Also, I forgot to mention quite important thing: I'm running WAMR inside Intel SGX enclave.

The good news is that I found a way to workaround using WAMR `1.1.2` with pinned `asmjit` commit (that would suggest that the problem is outside of my project). To workaround, I just used [`WAMR-1.1.2`][wamr-1.1.2] instead of [`master`](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main) (`git checkout tags/WAMR-1.1.2`) and applied cherry-picked patch https://github.com/bytecodealliance/wasm-micro-runtime/commit/578fbc5a55276576f3cc68742e31275e95535c2b that pins `asmjit` to this commit: https://github.com/asmjit/asmjit/commit/c1019f1642a588107148f64ba54584b0ae3ec8d1. I will try to pinpoint which commit from the WAMR repo breaks it.

[wamr-1.1.2]: https://github.com/bytecodealliance/wasm-micro-runtime/releases/tag/WAMR-1.1.2

TianlongLiang, 2023-03-21:
> I also forgot to mention a pretty crucial thing: I'm running WAMR in Intel SGX enclave.

😄That's okay, I figured it out with the compile options you used. One more thing I have to ask, are you using the SGX hardware mode or SGX simulation mode? I will try to reproduce this issue with the same environment.

TianlongLiang, 2023-03-23:
Meanwhile, can you try the latest version to see whether this issue persists? By the way, to locate exactly where to hang, maybe adding the CMake flag `-DCMAKE_BUILD_TYPE=Debug` can help the `strace` to obtain the source code level information.