wasm-micro-runtime/wasm-micro-runtime #2037: Compiling TypeScript to WebAssembly

xujuntwt95329, 2023-03-17:
# Compiling TypeScript to WebAssembly

WebAssembly lacks of application programming language, currently supported languages are mainly for system programming (C/C++/Rust). Some approaches embed a whole language runtime inside WebAssembly to support other languages (JS, python), but this is not friendly to footprint and performance. 
On the other hand, TypeScript, as a statically-typed super set of JavaScript, can likely run faster on WebAssembly if we make use of the type information during compilation. 

This issue is to track and discuss possible approaches to support compiling TypeScript to WebAssembly.

## Motivation
- Bring new programing language to WebAssembly to enhance the ecosystem
## Basic ideas
- Static compilation as much as possible
- Support dynamic through a builtin runtime library
- Leverage WebAssembly GC proposal
## Expected benefits
- Quick startup
- Small module size
- Friendly development experience to application developer
## Use case
- New programming language for IoT/FaaS/TEE
- Accelerator for web-based applications
## Dependent proposals
- [garbage collection (phase 3)](https://github.com/WebAssembly/gc)
	- For object presentation and memory management
- [type import (phase 1)](https://github.com/WebAssembly/proposal-type-imports)
	- For share type information between modules
- [stringref (phase 1)](https://github.com/WebAssembly/stringref)
	- For string encodings and interoperation between JavaScript and WebAssembly
- [component model (phase 1)](https://github.com/WebAssembly/component-model)
	- For module based distribution
- [exception handling (phase 3)](https://github.com/WebAssembly/exception-handling)
	- For supporting try-catch blocks in source code

## Experimental implementation
https://github.com/bytecodealliance/wasm-micro-runtime/pull/2034

## SIG
https://github.com/bytecodealliance/governance/pull/28/files

wenyongh, 2023-03-20:
The source code of [ts2wasm](https://github.com/bytecodealliance/wasm-micro-runtime/tree/ts2wasm_experiment/ts2wasm)  compiler is under branch `ts2wasm_experiment` of WAMR:
https://github.com/bytecodealliance/wasm-micro-runtime/tree/ts2wasm_experiment

Joneyao, 2023-03-22:
> 
请问一下 当前 ts2wasm 项目编译生成的wasm模块，只能通过v8引擎运行吗？ 其他常用的wasm 运行时无法运行？这个是什么原因？是因为当前的wasm内部有什么特殊规范吗？

Joneyao, 2023-03-22:
> 

[image]

xujuntwt95329, 2023-03-22:
@Joneyao  ts2wasm rely on the [WebAssembly GC proposal](https://github.com/WebAssembly/gc), we use latest version of [binaryen](https://github.com/WebAssembly/binaryen) to generate GC related opcodes, these opcodes only compatible to V8 version 11+, so currently we can only run the wasm module in these two approaches:
1. download latest V8 source code, and compile a `d8` executable, see [run_generated_module](https://github.com/bytecodealliance/wasm-micro-runtime/blob/ts2wasm_experiment/ts2wasm/tools/validate/module-run/run_generated_module.md) for more details
2. run in the browser (latest chrome is shipped with V8 version 11+), we will add some document on how to do this.

Note that the functionality of std libraries (like console.log) is under development, so currently the wasm module may lack of output capabilities, you can use chrome dev-tools to inspect the internal data (local/globals) inside the VM

goodcodedev, 2023-04-19:
Did you consider https://github.com/dudykr/stc as a possible base for typechecking? I don't know if it's possibly more mallable to the wasm case, but likely faster.

acomagu, 2023-04-27:
> The source code of [ts2wasm](https://github.com/bytecodealliance/wasm-micro-runtime/tree/ts2wasm_experiment/ts2wasm) compiler is under branch `ts2wasm_experiment` of WAMR: https://github.com/bytecodealliance/wasm-micro-runtime/tree/ts2wasm_experiment

This branch seems to be removed or renamed. Where did it go?