wasm-micro-runtime/wasm-micro-runtime #2054: [HELP] remote Debug linux wasm from windows lldb

xujiesh0510, 2023-03-24:
remote  Debug linux iwasm from linux version of lldb(https://github.com/bytecodealliance/wasm-micro-runtime/releases) is ok.

On windows follow the instruction(wasm-micro-runtime main branch):
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/source_debugging.md
git clone --branch release/13.x --depth=1 https://github.com/llvm/llvm-project
cd llvm-project
git apply ${WAMR_ROOT}/build-scripts/lldb-wasm.patch
mkdir build-lldb
cmake -S ./llvm -B build-lldb \
    -G Ninja \
    -DCMAKE_BUILD_TYPE=Release -DLLVM_ENABLE_PROJECTS="clang;lldb" \
    -DLLVM_TARGETS_TO_BUILD:STRING="X86;WebAssembly" \
    -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DLLVM_BUILD_BENCHMARKS:BOOL=OFF \
    -DLLVM_BUILD_DOCS:BOOL=OFF  -DLLVM_BUILD_EXAMPLES:BOOL=OFF  \
    -DLLVM_BUILD_LLVM_DYLIB:BOOL=OFF  -DLLVM_BUILD_TESTS:BOOL=OFF  \
    -DLLVM_ENABLE_BINDINGS:BOOL=OFF  -DLLVM_INCLUDE_BENCHMARKS:BOOL=OFF  \
    -DLLVM_INCLUDE_DOCS:BOOL=OFF  -DLLVM_INCLUDE_EXAMPLES:BOOL=OFF  \
    -DLLVM_INCLUDE_TESTS:BOOL=OFF -DLLVM_ENABLE_LIBXML2:BOOL=ON
cmake --build build-lldb --target lldb --parallel 8

(lldb) platform select remote-linux
(lldb) process connect -p wasm connect://127.0.0.1:1234

llldb.exe can connect to iwasm in linux ,but breakpoint has no effects

(lldb) platform select remote-linux
  Platform: remote-linux
 Connected: no
(lldb) process connect -p wasm connect://192.168.139.134:1234
Process 1 stopped
* thread #1, name = 'nobody', stop reason = trace
    frame #0: 0x40000000000001e6
->  0x40000000000001e6: block
    0x40000000000001e8: call   20
    0x40000000000001ee: local.tee 0
    0x40000000000001f0: i32.eqz
(lldb) b foo
Breakpoint 1: no locations (pending).
WARNING:  Unable to resolve breakpoint to any actual locations.

how to slove this ?

xujuntwt95329, 2023-03-30:
Hi, previously I have built a Windows version, you can have a try: 
[debug-windows.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/11108164/debug-windows.zip)

As for the building instructions, I need to find a Windows environment to verify the steps, I‘ll update later

xujiesh0510, 2023-03-30:
Thank you very much for your help. It took me days trying  to compile lldb on windows.

xujuntwt95329, 2023-03-30:
You are welcome, hope the prebuilt binary can help you. 
It's really difficult to build on windows, and I found my previous steps don't work on latest windows, I need some time to figure out the problem

closed 2023-04-19 by wenyongh