wasm-micro-runtime/wasm-micro-runtime #2055: [question] How to build iwasm for ARM64 windows?

ikorin24, 2023-03-24:
I want to build iwasm for ARM64 windows, but I don't know how to do it. Is there any docs with instructions?
I looked at the [README](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/product-mini/README.md#windows), but it doesn't mention ARM64 windows.

Is it possible to build iwasm for ARM64 windows on x86-64 windows?

I tried the following:

I used `WAMR-1.1.2` tagged commit. (fb8727ba683b3969cc6075356bf109defefe6003)

1. Change dir
```
$ cd product-mini/platforms/windows/
```

2. Add a line to `CMakeLists.txt`
```
set (WAMR_BUILD_TARGET "AARCH64")
```
3. Build it (on x64 PC)
```
$ mkdir build
$ cd build
$ cmake ..
$ cmake --build . --config Release
```

error
```
wasm_runtime_common.obj : error LNK2001: unresolved external symbol "invokeNative"  [<my-working-dir>\wasm-micro-runtime\product-mini\platforms\windows\build\libiwasm.vcxproj]
```

xujiesh0510, 2023-03-24:
It seems you need to specify the right DCMAKE_TOOLCHAIN_FILE, and you need cross-compile-toolchain
[image]

[image]

wenyongh, 2023-03-31:
Hi, it seems that the `invokeNative_xxx.s` or `invokeNative_xxx.asm` file isn't compiled and linked, could you try modifying `iwasm_common.cmake` to add it by yourself?
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/common/iwasm_common.cmake#L42-L50

ikorin24, 2023-04-23:
I could not figure out how to rewrite `iwasm_common.cmake`.
However, I wasn't actually calling the process in iwasmNative_xxx.asm, just the inability to build.

So, by setting `WAMR_BUILD_INVOKE_NATIVE_GENERAL` to 1, I was able to build iwasm using `invokeNative_general.c` instead. This is incomplete and does not work well, but since I do not invoke that code it was not a problem for me.

https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/common/iwasm_common.cmake#L28-L37

I closed this issue. Thank you

closed 2023-04-23 by ikorin24