wasm-micro-runtime/wasm-micro-runtime #2063: On WAMR-1.2.0's pre-release status

ZhongRuoyu, 2023-03-25:
Hi! [Homebrew](https://github.com/Homebrew/brew) maintainer here. While packaging WAMR 1.2.0, I noticed that it is currently marked as a pre-release on the [releases page](https://github.com/bytecodealliance/wasm-micro-runtime/tree/WAMR-1.2.0). So, I'm filing this issue to confirm what that means. Is it not ready for distribution yet?

wenyongh, 2023-03-26:
Hi, the distribution is basically ready with many tests done, but we hope to do more tests and update some documents (if needed), which requires more time. And some developers may hope to release a version for their internal use, so we create a pre-release first. Refer to: https://github.com/bytecodealliance/wasm-micro-runtime/pull/2052

ZhongRuoyu, 2023-03-26:
Hi @wenyongh, thanks for your clarification! In that case, let me postpone shipping 1.2.0 until it's fully ready for general usage.

wenyongh, 2023-04-07:
@ZhongRuoyu WAMR-1.2.1 was released: https://github.com/bytecodealliance/wasm-micro-runtime/releases/tag/WAMR-1.2.1, please have a try, thanks.

ZhongRuoyu, 2023-04-07:
Thanks @wenyongh. It seems someone did this before me: #127717. Now release 1.2.1 is on Homebrew.

closed 2023-04-07 by ZhongRuoyu