wasm-micro-runtime/wasm-micro-runtime #2065: WAMR profiling support

eloparco, 2023-03-26:
I'm creating this issue to track the implementation of profiling support in WAMR. 

The goal is to understand which high-level (e.g. C, Rust) functions take most of the CPU time. To do that, we would like to use the `perf` Linux tool, by making WAMR generate the required symbol translation information, as described in https://www.brendangregg.com/perf.html (`4.3. JIT Symbols` chapter).

I'll update the issue once the plan and timeline get more defined. Feedback is appreciated.

lum1n0us, 2023-08-09:
@eloparco Any updates on this idea? 

There is another requirement that may involve `perf` supporting. It would be very helpful if we can share something.

eloparco, 2023-08-20:
Hey @lum1n0us, we did some research and were thinking to use [perf-maps](https://github.com/torvalds/linux/blob/master/tools/perf/Documentation/jit-interface.txt) as it's done in Wasmtime. But then the project got deprioritized and currently we're not planning on investing time in it.

loganek, 2023-08-23:
@lum1n0us would you be able to share the other usecase for the perf support?

lum1n0us, 2023-08-24:
Yeah. They intend to use [flame graph](https://www.brendangregg.com/FlameGraphs/cpuflamegraphs.html) to do workloads analysis.

eloparco, 2024-02-23:
Closing the issue since it's now implemented (https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/perf_tune.md?plain=1#L96)

closed 2024-02-23 by eloparco