wasm-micro-runtime/wasm-micro-runtime #2067: [Question] What will trigger the WAMR to do the tier-up?

hungryzzz, 2023-03-26:
Hi, I read the description from WAMR's document, **_"The WAMR supports two JIT tiers - Fast JIT, LLVM JIT, and dynamic tier-up from Fast JIT to LLVM JIT."_**. 

So what will trigger the WAMR to do the tier-up, from the Fast JIT to LLVM JIT? And are there something difference between these two JITs? Or will the runtime collect some profiling information during the first tier execution?

Thank you!

Kevin0626, 2023-03-26:
@hungryzzz , there are more details in the blog below:
- https://bytecodealliance.github.io/wamr.dev/blog/introduction-to-wamr-running-modes/

Please see if it can give you enough information.

hungryzzz, 2023-03-28:
Thank you, the blog introduce the basic information of WAMR very clearly, but I did't find the answers of my above questions. Maybe my description of the problems was not clear enough, which caused you to misunderstand my meaning. Let me repeat it.

In the multi-tier-JIT mode, what will trigger the Fast JIT tier up to the LLVM JIT? Does it depend on some profiling information collected during Fast JIT execution? Or will the LLVM JIT be triggered anyway after the Fast JIT finishes, and the tiering up just happens when the LLVM JIT finishes?

Thanks again for answering my questions!

wenyongh, 2023-03-28:
Hi, when Fast JIT and LLVM JIT are both enabled, and JIT lazy mode is enabled (which is enabled by default), the multi-tier jit will be enabled automatically. Please build iwasm with the commands below:
```bash
cd product-mini/platforms/linux
mkdir build && cd build
cmake .. -DWAMR_BUILD_FAST_JIT=1 -DWAMR_BUILD_JIT=1
make -j
```
By default iwasm enables multi-tier-jit by default, and you can run with other running modes by `iwasm --interp`, `iwasm --fast-jit`, `iwasm --llvm-jit`.

The Multi-tier JIT firstly sets the Fast JIT func ptrs as `compile itself and then run` and starts to run wasm app with Fast JIT mode, and at the same time runtime creates backend threads to compile all fast jit functions, so the fast jit code will be called as soon as possible to get a quick cold startup. And then the backend threads continue to compile the llvm jit functions, and as long as an llvm jit function is compiled, the calling of this wasm function will be switched from fast jit to llvm jit, by this way the fast jit will be switched to llvm jit gradually (and as soon as possible), so as to achieve best performance.

hungryzzz, 2023-03-29:
Okay, I got it, it's really clear! Thank you very much!

closed 2023-03-29 by hungryzzz