wasm-micro-runtime/wasm-micro-runtime #2075: lldb remote debug , can not connect to  embedders

xujiesh0510, 2023-03-28:
ENV:
TWO LINUX machine, one for lldb ,one for  embedder server .

lldb remote debug, lldb can not connect to  embedder server

embedder server cmakeList.txt:

cmake_minimum_required(VERSION 3.10)
project(HELLO C)
MESSAGE(STATUS  "${PROJECT_SOURCE_DIR} ${PROJECT_BINARY_DIR}" )
set(CMAKE_C_STANDARD 11)
set (WAMR_BUILD_PLATFORM "linux")
set (WAMR_BUILD_TARGET "X86_64")
set (WAMR_BUILD_INTERP 1)
set (WAMR_BUILD_DEBUG_INTERP 1)
set (WAMR_BUILD_AOT 0)
set (WAMR_BUILD_LIBC_BUILTIN 1)
set (WAMR_BUILD_LIBC_WASI 1)
set (WAMR_ROOT_DIR "/mnt/e/Study/C/wasm-micro-runtime-WAMR-1.2.0")
include (${PROJECT_SOURCE_DIR}/../wasm-micro-runtime-WAMR-1.2.0/build-scripts/runtime_lib.cmake)
add_library(vmlib ${WAMR_RUNTIME_LIB_SOURCE})
add_executable(hello   src/main.c )
target_link_libraries (hello vmlib  -lpthread -lm)

main.c

#include <stdio.h>
#include <string.h>
#include "wasm_export.h"
#include "platform_common.h"

extern int fib(int n);
int main()
{
    char * x = NULL;
    char *buffer, error_buf[128];
    wasm_module_t module;
    wasm_module_inst_t module_inst;
    wasm_function_inst_t func;
    wasm_exec_env_t exec_env;
    uint32 size;

    wasm_runtime_init();

    /* Open up card file */
    FILE *file = fopen("/mnt/e/Study/C/webassembly/lib/main.wasm", "r");

    if (!file) {
        fprintf(stderr, "Unable to open/create file\n");
        return 1;
    }

    /* Size of file */
    fseek(file, 0, SEEK_END);
    size = ftell(file);
    fseek(file, 0, SEEK_SET);

    if (!(buffer = malloc(size * sizeof(char)))) {
        fprintf(stderr, "Failed to allocate memory\n");
        return 1;
    }

    size_t read = fread(buffer, 1, size, file);

/* all the runtime memory allocations are retricted in the global_heap_buf array */
    static char global_heap_buf[512 * 1024*8];
    RuntimeInitArgs init_args;
    memset(&init_args, 0, sizeof(RuntimeInitArgs));
    strcpy(init_args.ip_addr, "172.24.235.192");
    init_args.instance_port = 1234;
/* configure the memory allocator for the runtime */
    init_args.mem_alloc_type = Alloc_With_Pool;
    init_args.mem_alloc_option.pool.heap_buf = global_heap_buf;
    init_args.mem_alloc_option.pool.heap_size = sizeof(global_heap_buf);

    if (!wasm_runtime_full_init(&init_args)) {
        return false;
    }

    /* parse the WASM file from buffer and create a WASM module */
    module = wasm_runtime_load(buffer, size, error_buf, sizeof(error_buf));
    uint32 stack_size = 80920*2, heap_size = 80920*2;
    /* create an instance of the WASM module (WASM linear memory is ready) */
    module_inst = wasm_runtime_instantiate(module, stack_size, heap_size,
                                           error_buf, sizeof(error_buf));

    /* creat an execution environment to execute the WASM functions */
    exec_env = wasm_runtime_create_exec_env(module_inst, stack_size);
    wasm_runtime_start_debug_instance(exec_env);

    int xz =0 ;
    scanf("%d",&xz);
    printf("good .......");
}

xujuntwt95329, 2023-03-30:
Hi, I can't re-produce the problem on my machine, could you please run iwasm with `-v=5` and attach the log here?

xujiesh0510, 2023-03-30:
> Hi, I can't re-produce the problem on my machine, could you please run iwasm with `-v=5` and attach the log here?

Sorry. A bit confused here. I'm trying to embed wasm debugger  into my program ,act as a debugger server. 
in client side, i want to user lldb remote conect to debugger server and debug the wasm loaded in the debugger server.
i can not connect to the  debugger server using lldb.
I've test the the same [main.wasm.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/11086818/main.wasm.zip) . lldb connect to  iwasm is ok. 
It seems that ' run iwasm with -v=5'  does not help too much . debugger server does not need iwasm.exe

xujiesh0510, 2023-03-30:
HI, thanks for your time.
this is the log info:

[06:57:10:496 - 7F0A68C8A880]: No aux stack info for this module, can't create thread
[06:57:10:612 - 7F0A68C8A880]: No aux stack info for this module, can't create thread
[06:57:10:938 - 7F0A68C89700]: control thread of debug object 0x563b446a11e8 start

[06:57:10:311 - 7F0A68C89700]: Debug server listening on 0.0.0.0:1234

[06:57:10:362 - 7F0A68C89700]: listen for gdb client
[06:57:10:402 - 7F0A68C89700]: waiting for gdb client to connect...
[06:59:18:993 - 7F0A68C89700]: accept gdb client
[06:59:18:655 - 7F0A68C89700]: receive request:Q StartNoAckMode

[06:59:18:699 - 7F0A68C89700]: handle_general_set:StartNoAckMode

[06:59:18:704 - 7F0A68C89700]: send replay:OK
[06:59:18:767 - 7F0A68C89700]: receive request:q Supported:xmlRegisters=i386,arm,mips,arc;multiprocess+

[06:59:18:812 - 7F0A68C89700]: handle_general_query:Supported

[06:59:18:817 - 7F0A68C89700]: send replay:qXfer:libraries:read+;PacketSize=1000;
[06:59:18:230 - 7F0A68C89700]: receive request:Q ThreadSuffixSupported

[06:59:18:288 - 7F0A68C89700]: handle_general_set:ThreadSuffixSupported

[06:59:18:293 - 7F0A68C89700]: send replay:
[06:59:18:597 - 7F0A68C89700]: receive request:Q ListThreadsInStopReply

[06:59:18:642 - 7F0A68C89700]: handle_general_set:ListThreadsInStopReply

[06:59:18:647 - 7F0A68C89700]: send replay:
[06:59:18:338 - 7F0A68C89700]: receive request:q HostInfo

[06:59:18:385 - 7F0A68C89700]: handle_general_query:HostInfo

[06:59:18:391 - 7F0A68C89700]: send replay:vendor:wamr;ostype:wasi;arch:wasm32;triple:7761736d33322d77616d722d776173692d7761736d;endian:little;ptrsize:4;
[06:59:19:395 - 7F0A68C89700]: receive request:v Cont?

[06:59:19:460 - 7F0A68C89700]: handle_v_packet:Cont?

[06:59:19:464 - 7F0A68C89700]: send replay:vCont;c;C;s;S;
[06:59:19:039 - 7F0A68C89700]: receive request:q VAttachOrWaitSupported

[06:59:19:098 - 7F0A68C89700]: handle_general_query:VAttachOrWaitSupported

[06:59:19:103 - 7F0A68C89700]: send replay:
[06:59:19:571 - 7F0A68C89700]: receive request:Q EnableErrorStrings

[06:59:19:614 - 7F0A68C89700]: handle_general_set:EnableErrorStrings

[06:59:19:618 - 7F0A68C89700]: send replay:OK
[06:59:19:543 - 7F0A68C89700]: receive request:q ProcessInfo

[06:59:19:587 - 7F0A68C89700]: handle_general_query:ProcessInfo

[06:59:19:593 - 7F0A68C89700]: send replay:pid:1;parent-pid:1;vendor:wamr;ostype:wasi;arch:wasm32;triple:7761736d33322d77616d722d776173692d7761736d;endian:little;ptrsize:4;
[06:59:19:977 - 7F0A68C89700]: receive request:?

xujuntwt95329, 2023-03-30:
Hi, seems the debug-server is hang at https://github.com/bytecodealliance/wasm-micro-runtime/blob/403ccf02b72760888b5cf7f8bab69723dad69a8e/core/iwasm/libraries/debug-engine/handler.c#L524

Are you using `iwasm` or your own embedder? I haven't seen such problem with `iwasm`, if you are using your own embedder, then could you tell me what's the status of the interpreter thread? Seems it's already running and didn't stopped after debugger connected

closed 2023-03-31 by xujiesh0510