wasm-micro-runtime/wasm-micro-runtime #2093: Unexpected failed execution: ref.func 0

erxiaozhou, 2023-04-01:
# Build commands
I compile the code with commit id aaf671d688592784d1f8e000531cd98c2c22bfac.
Platform: Ubuntu 20.04
CPU: amd64 
compile:
```shell
export CC=/usr/bin/clang-14
export CXX=/usr/bin/clang++-14
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_REF_TYPES=1 -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Debug -DWAMR_BUILD_FAST_INTERP=1 -DWASM_ENABLE_LIB_WASI_THREADS=1 -Bbuild
cd build
make -j40
```
execute:
```shell
iwasm --heap-size=0 -f to_test <tc_name>
```

test case:

[iwasm_2.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/11129564/iwasm_2.zip)

# Actual behavior:
An exception:
```
WASM module load failed: undeclared function reference
```
# Expected behavior:
The test case is executable

wenyongh, 2023-04-03:
Hi, thanks for reporting the issue! I submitted PR #2099 to fix it, could you please try again?

erxiaozhou, 2023-04-03:
> Hi, thanks for reporting the issue! I submitted PR #2099 to fix it, could you please try again?

Thank you for your nice reply. it can work now!

closed 2023-04-24 by wenyongh