wasm-micro-runtime/wasm-micro-runtime #2094: Unexpected failed execution

erxiaozhou, 2023-04-01:
# Build commands
I compile the code with commit id aaf671d688592784d1f8e000531cd98c2c22bfac.
Platform: Ubuntu 20.04
CPU: amd64 
compile:
```shell
export CC=/usr/bin/clang-14
export CXX=/usr/bin/clang++-14
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_REF_TYPES=1 -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Debug -DWAMR_BUILD_FAST_INTERP=1 -DWASM_ENABLE_LIB_WASI_THREADS=1 -Bbuild
cd build
make -j40
```
execute:
```shell
iwasm --heap-size=0 -f to_test <tc_name>
```

test case:

[iwasm_1.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/11129567/iwasm_1.zip)

# Actual behavior:
An exception:
```
WASM module load failed: a module with WASI apis must export memory by default
```
# Expected behavior:
The test case is executable

lum1n0us, 2023-04-03:
May I ask how generate the test .wasm? It doesn't seems like a standard wasm32-wasi module. In that case, it is better to disable WASI-LIBC and use BUILTIN-LIBC by `-DWAMR_BUILD_LIBC_WASI=0 -DWAMR_BUILD_LIBC_BUILTIN=1`.

Plus, it always requires a `main()` function in the original source file.

lum1n0us, 2023-04-03:
Might want to refer this https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wasm_app.md#build-wasm-applications-with-wasi-sdk, if trying to compile C to Wasm

erxiaozhou, 2023-04-03:
I get the root cause, thank you!

erxiaozhou, 2023-04-03:
> Might want to refer this https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wasm_app.md#build-wasm-applications-with-wasi-sdk, if trying to compile C to Wasm

I get the root cause, thank you!