wasm-micro-runtime/wasm-micro-runtime #2096: Unexpected execution

erxiaozhou, 2023-04-02:
# Build commands
I compile the code with commit id aaf671d688592784d1f8e000531cd98c2c22bfac.
Platform: Ubuntu 20.04
CPU: amd64 
compile:
```shell
export CC=/usr/bin/clang-14
export CXX=/usr/bin/clang++-14
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_REF_TYPES=1 -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Debug -DWAMR_BUILD_FAST_INTERP=1 -Bbuild
cd build
make -j40
```
execute:
```shell
iwasm --heap-size=0 -f to_test <tc_name>
```

[iwasm_table_OOB.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/11132202/iwasm_table_OOB.zip)

# Actual behavior:
No exception

# Expected behavior:
An exception indicating that "out of bounds table access"

wenyongh, 2023-04-03:
Hi, the opcode `table.init` in this case takes the operand n, src, dst with 0, 0, -2147483648, though the destination is "out of bounds", the size to copy is 0, per my understanding, it should be ok to just ignore the initialization.

erxiaozhou, 2023-04-03:
> Hi, the opcode `table.init` in this case takes the operand n, src, dst with 0, 0, -2147483648, though the destination is "out of bounds", the size to copy is 0, per my understanding, it should be ok to just ignore the initialization.

Thank you for your nice reply!

erxiaozhou, 2023-04-16:
> Hi, the opcode `table.init` in this case takes the operand n, src, dst with 0, 0, -2147483648, though the destination is "out of bounds", the size to copy is 0, per my understanding, it should be ok to just ignore the initialization.

Hi~.   Maybe it is better to fix this "bug". 
According to the specification, there should be a trap, (lines 16, 17; the runtime should first check whether d+n greater than the length of the table and then check whether n equals 0).  Furthermore, I have tried this test case on other runtimes and they all failed due to "Table OOB".  So I think fixing this behavior might be a better option. Thank you~

[image]

yamt, 2023-04-17:
i agree it should trap.
cf. https://github.com/WebAssembly/spec/blob/2a992ec2d30b0a7f0533ebdb5fe0cd7eb42553ca/test/core/table_init.wast#L671-L693