wasm-micro-runtime/wasm-micro-runtime #2097: About start function name

erxiaozhou, 2023-04-02:
# Build commands
I compile the code with commit id aaf671d688592784d1f8e000531cd98c2c22bfac.
Platform: Ubuntu 20.04
CPU: amd64 
compile:
```shell
export CC=/usr/bin/clang-14
export CXX=/usr/bin/clang++-14
cd product-mini/platforms/linux/;rm -rf build
cmake  -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_REF_TYPES=1 -DWAMR_BUILD_AOT=0 -DCMAKE_BUILD_TYPE=Debug -DWAMR_BUILD_FAST_INTERP=1 -DWASM_ENABLE_LIB_WASI_THREADS=1 -Bbuild
cd build
make -j40
```
execute:
```shell
iwasm --heap-size=0 -f to_test <tc_name>
```
xxxxx

# Actual behavior:
An exception
```
WASM module load failed: the signature of builtin _start function is wrong
```

# Expected behavior:
Output: 0x80000001:i32

Thank you!

lum1n0us, 2023-04-03:
like the `main()` for a C program, Wasm modules require a specific function with specific signature as the entry point of execution. Luck for us, compilers will give us a hint.

https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wasm_app.md#build-wasm-applications-with-wasi-sdk is a quick guide for how to compile C to Wasm. Hope it helpful.

erxiaozhou, 2023-04-03:
> like the `main()` for a C program, Wasm modules require a specific function with a specific signature as the entry point of execution. Luck for us, compilers will give us a hint.
> 
> https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/build_wasm_app.md#build-wasm-applications-with-wasi-sdk is a quick guide for how to compile C to Wasm. Hope it helpful.

[iwasm_name_not_sure.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/11135171/iwasm_name_not_sure.zip)

I have uploaded the .wasm file; In this test case, the entry function has the name "to_test" and the name "_start".  When invoking the test case by command `iwasm --heap-size=0 -f to_test`, the entry function should be called. By the way, this command is executable on version 6820af621265413b677e450cf126bb66c9f77077, so I think it is a bug.

Here is another interesting test case. In the example below, the name of the entry function is only specified as "to_test" and it is executable. 
[tt.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/11135174/tt.zip)

erxiaozhou, 2023-04-03:
Maybe I need to re-compile the project and try again ...

erxiaozhou, 2023-04-03:
I get the root cause, thank you!

yamt, 2023-04-07:
it seems like a bug in the loader.
when `check_wasi_abi_compatibility` failed, wamr should just treat it as non-wasi module, rather than bailing out.

lum1n0us, 2023-04-07:
IMO, `WAMR_BUILD_LIBC_WASI` is kind of a switcher to reshape *iwasm* and sets up implicit assumptions for the running environment. like preparing WASI context, expecting `_initializer()` or `_start`, and so on.

another reason of using a *faill-fast* strategy is to give users a clear feedback that `wasm32-unknown` is different with `wasm32-wasi`. 

Plus, Compilers of C/C++, Go, Rust are able to generate `wasm32-wasi` easily. We tend to encourage people use wasm32-wasi.

lum1n0us, 2023-04-16:
> but rejecting non-wasi modules is too much, IMO.

IIUC, it is not rejecting non-wasi module. It is about using the right compilation option group. 

- If want to run wasi modules, using `-DWAMR_BUILD_LIBC_WASI=1`
- If want to run no-wasi modules, using `-DWAMR_BUILD_LIBC_BUILTIN=1`

yamt, 2023-05-10:
> > but rejecting non-wasi modules is too much, IMO.
> 
> IIUC, it is not rejecting non-wasi module. It is about using the right compilation option group.
> 
>     * If want to run wasi modules, using `-DWAMR_BUILD_LIBC_WASI=1`
> 
>     * If want to run no-wasi modules, using `-DWAMR_BUILD_LIBC_BUILTIN=1`

i meant that it was too much to make `-DWAMR_BUILD_LIBC_WASI=1` reject non-wasi modules.

lum1n0us, 2023-05-10:
We used to meet some non-wasi modules includes `_initialize()` and `_start()`. It is hard to tell those special named function should be executed? should be executed multiple times or only once?

Therefore, we made a simple decision.