wasm-micro-runtime/wasm-micro-runtime #2123: Segfault while compile C++ wasm code using wamrc built with debugging AOT enabled

acane77, 2023-04-12:
Segfault while compile C++ wasm code using wamrc built with debugging AOT enabled.

I compile wamrc with the commands described in [this link](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/doc/source_debugging.md), the debugging with AOT section.

Here is a small sample code. 

```c++
#include <cstdio>
#include <map>
#include <vector>

int main(int argc, const char** argv) {
    std::map<int, int> m;
    m[1] = 1;
    std::vector<int> x = {1,2,3};
    printf("x[2] = %d", x[2]);
    int s = 0;
    for (int i=0; i <= 10; i++) {
        s += i;
    }
    printf("sum=%d\n", s);
    return 0;
}

```

I compile this source code with wasi-sdk using the following command
```
 /opt/wasi-sdk-16.0/bin/clang++ test.cpp -g -O0 -std=c++11 \
                                    -fno-exceptions \
                                    -Wl,--export=__heap_base -Wl,--export=__data_end -Wl,--export=free -Wl,--export=malloc -Wl,--allow-undefined\
                                    -o test.wasm
```

and compile the wasm to AOT with warmc:
```
wamrc --size-level=0 -o test.aot test.wasm 
```
and wamrc crashed with segmentation fault.

the outputs are

```
Create AoT compiler with:
  target:        x86_64
  target cpu:    skylake
  cpu features:  
  opt level:     3
  size level:    0
  output format: AoT file
[03:08:37:655 - 7FDF3C7F5C00]: function args number dismatch!:value number=1, function args=0
[03:08:37:089 - 7FDF3C7F5C00]: function args number dismatch!:value number=2, function args=1
[03:08:37:597 - 7FDF3C7F5C00]: function args number dismatch!:value number=2, function args=1
[03:08:37:878 - 7FDF3C7F5C00]: function args number dismatch!:value number=2, function args=1
[03:08:37:519 - 7FDF3C7F5C00]: function args number dismatch!:value number=1, function args=0
[03:08:37:593 - 7FDF3C7F5C00]: function args number dismatch!:value number=1, function args=0
[03:08:37:388 - 7FDF3C7F5C00]: function args number dismatch!:value number=2, function args=1
Segmentation fault (core dumped)
```

This problem only occurs when compiling code which invokes C++ standard libraries (for example, iostream, STL libraries). 
Compiling codes which only call C libraries are OK, (pure C code is also OK)

TianlongLiang, 2023-04-19:
Thanks for bringing this issue to our attention. I will try your code and see if I can help; I will come back to you ASAP

TianlongLiang, 2023-05-17:
Sorry for the late reply. I have looked into this issue. Currently, AOT debugging is experimental and only has limited functionality. It aimed to support C first. It hardly covers any additional C++ features for now. The current function-to-function debug symbol mapping only supports basic functions, meaning it wouldn't work on a function incompatible with C(member function, template function, lambda function, etc). We may support it in the future, but unfortunately, it could be a while. You are more than welcome to use Interpreter debugging feature, it should support more functionality you need