wasm-micro-runtime/wasm-micro-runtime #2124: wamrc is too slow for some wasm modules

yamt, 2023-04-13:
wamrc (especially its optimization pass) takes hours on my environment.

wasm module: [many_stack7.wasm.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/11218543/many_stack7.wasm.zip)
its source: https://github.com/yamt/toywasm/blob/f917a189f3970959fb5f2ea72083cc62b88829c6/wat/many_stack.wat.in#L4

if it matters, it's amd64 macos wamrc binary from 1.2.1 release assets.

yamt, 2023-04-13:
`iwasm --llvm-jit` seems to have the same problem.

wenyongh, 2023-04-17:
Normally the compilation of `iwasm --llvm-jit` should be much faster than wamrc: (1) it supports lazy compilation in the main thread and only needs to compile the wasm function to call when the function hasn't been compiled, (2) and it supports parallel compilation, there are 4 backend threads compiling at the same time by default.

Besides, for linux/macos x86-64, the multi-tier jit is supported, the fast-jit is launched to compile and run quickly, and when llvm-jit functions are compiled, the fast-jit will be switched to llvm-jit as soon as possible.

It is not easy to reduce the compilation time of wamrc since unlike llvm jit mode, the llvm non-jit mode doesn't support parallel compilation yet.

yamt, 2023-04-18:
for this particular module, which is basically a single function, i guess `--llvm-jit` is slow as `wamrc`.
multi-tier jit certainly helps.