wasm-micro-runtime/wasm-micro-runtime #2126: Component model support

steren, 2023-04-14:
WASI has moved to preview 2, which is based on the [Component Model](https://github.com/WebAssembly/component-model).

In a [recent community presentation](https://www.youtube.com/watch?v=9pLa7PUhPYA), it was briefly mentioned that the component model would be in WAMR's roadmap, but I wasn't able to find an issue or more written information about this topic. I suggest using this issue as a feature request and as a way to attach any existing links to plans and timelines for this topic.

yamt, 2023-05-08:
technically, preview2 does not require component model.
i agree it's nice to have component model though.

btw, to make component model useful, i suspect we need to align our import/export model to the standard semantics (cf. https://github.com/bytecodealliance/wasm-micro-runtime/issues/1353)
because many of the current use cases of component model involves import/export heavily.

lum1n0us, 2023-05-08:
> btw, to make component model useful, i suspect we need to align our import/export model to the standard semantics 

Yes, need fully alignment firstly.

squillace, 2023-08-13:
I'm extremely interested in preview 2 and component model support in wamr for our usage. Love to hear thinking on how to help this out with wamr.

lum1n0us, 2023-08-24:
@squillace  I am not sure about that. But would you mind sharing your usage about *preview2* and *component model*. Since all I've heard about it are kinds of concern, and you are the first one to stand for it.

nilslice, 2023-12-05:
+1 on interest in @squillace use case! happy to try to find ways to support via contribution here.

trcrsired, 2024-02-09:
how does preview2 work without component model? How does it know whether it is 32bit wasm or 64bit wasm?

weiyuemin, 2024-11-15:
Any update on this? It sounds great to have component model in wamr

lum1n0us, 2024-11-19:
There isn't a set timetable for the component model at this time. However, we are getting ready for it.  Like by restructuring the code base through the implementation of instantiation linking. We are open to any advanced plans on how to effectively support the component model and how to correctly integrate the component model into the embedded world.

GordonSmith, 2025-03-10:
Not sure if it is of interest, but I started work on a C++ ABI implementation of the component model here:  https://github.com/GordonSmith/component-model-cpp 
I am trying to keep it runtime agnostic and am using it as an excuse to get used to some of the newer C++language features...

lum1n0us, 2025-03-12:
Excellent. I'm looking forward to seeing an example.

GordonSmith, 2025-03-19:
Just working through some initial samples - plenty of work todo, but hopefully you can see the use pattern take shape:
* wit:  https://github.com/GordonSmith/component-model-cpp/blob/trunk/samples/wasm/sample.wit
* wamr:  https://github.com/GordonSmith/component-model-cpp/blob/trunk/samples/wamr/main.cpp#L171-L265

rinthnoid, 2025-09-19:
Is there any progress towards WAMR wasip2/component model since it is interesting feature yet to come in other runtimes except only wasmtime. And also the wasmtime is very hard to compile for android operating system.
It would be far more than nice if WAMR able support the component model wasm modules.

oberstet, 2025-12-28:
rgd https://github.com/bytecodealliance/wasm-micro-runtime/tree/dev/cm_wasip2

Thanks for sharing this update! — this is exciting work and exactly what the embedded Wasm community needs.

I'm very interested in this from an MCU perspective. My target environment is *Zephyr on devices with ~512KB RAM*, where WAMR is currently the only viable runtime option. The lack of Component Model / WASIp2 support has been a significant gap, as it forces us to stay on Preview 1 while the broader ecosystem moves forward.

Beyond the technical benefits of the Component Model (explicit interfaces, composability, capability-based security), I see strong alignment with **SLSA Level 4** supply chain security requirements:

- **Hermetic builds**: Components have fully declared dependencies via WIT — no ambient authority or hidden imports
- **Provenance tracking**: Components are self-describing artifacts that can be signed, verified, and attested
- **Isolation guarantees**: The sandboxing model limits blast radius even if a component is compromised

For regulated industries (aerospace, automotive, medical, industrial) running on constrained hardware, this combination of Component Model + MCU-capable runtime could be a game-changer! I guess this won't be anything new for you, just my 2cts and positive feedback;)

rodrigoflx-denso, 2026-10-08:
> My understanding is that this is one of the single largest ever code contributions to the WAMR project. My quick `cloc` analysis shows that this contribution represents 25% jump in the lines of code in the project.
> 
> Maintaining such a large code contribution for any project would be a challenge. I was wondering
> 
> 1. Is there an offer from Airbus or others to help provide support for the new code contributions?
> 2. Genuine and related question - what's the user demand for the component model with WAMR?
>    
>    1. (and) would this translate into maintenance support?
> 
> **Background** Looking at WAMR's recent commit history, it looks like there are maybe 2 - 3 individuals providing the bulk of support (bug & security tickets, etc.) at the moment - hence the questions about support, and the jump in project size. As with all things open source, once the new code is widely distributed and folks start testing it out, you could imagine there would be a wave of security issues / bugs / edge cases found. It would be important that the project would have the expertise of folks familiar with this addition to be able to provide the necessary support. This would mean responding to support tickets within a given number of hours / days, etc.
> 
> Seeing [@oberstet](https://github.com/oberstet) 's excitement around the contribution - which is awesome. Is there an actual deployment in mind using the WAMR + Component Model, where, say, WasmTime wouldn't currently work?
> 
> It is interesting to observe that this is the first attempt to add component model support to WAMR. The only other full runtime which has support for the component model is WasmTime. My understanding is that there hasn't been a significant heavy user of WAMR who has expressed a need for the Component Model (yet). If there was a huge demand that could be translated into maintenance support, that would be awesome.

At DENSO Automotive, an usecase that is being currently explored is embedding WAMR into OP-TEE (or any similar TEE that implement PSA/CCA ARM archs) for enabling dynamic secure applications. Although a specific OEM hasn't been narrowed down, we are expecting some Qualcomm platform. wasmtime wouldn't be a fit due to the constrained nature of the supported API (GP API) as well as the implementation (OP-TEE).

Support for the component model would be important for our usecase specially due to the capability-based security the component model brings along (if that would be part of Airbus's contrib until now).