wasm-micro-runtime/wasm-micro-runtime #2129: release process is not documented

yamt, 2023-04-18:
i wanted to test my changes to github workflow files by making a release on my fork of this repo.
but i couldn't find any doc explaining the procedure. (eg. how to kick `release_process.yml`)

yamt, 2023-04-18:
my current understanding is

1. update version.h (and release notes) push it
2. do something like ```gh workflow --repo yamt/wasm-micro-runtime run --ref test-release release_process.yml -f require_confirmation=true```

lum1n0us, 2023-04-19:
will not create a release branch. *release_process.yml* will create a tag based on the content of *version.h*. So `--ref` may not necessary. 

BTW, `require_confirmation` leads to a *github pre-release*.

yamt, 2023-04-20:
> will not create a release branch. _release_process.yml_ will create a tag based on the content of _version.h_. So `--ref` may not necessary.
> 
> BTW, `require_confirmation` leads to a _github pre-release_.

i used `--ref` just because i tested it on a branch of my fork.