wasm-micro-runtime/wasm-micro-runtime #2136: vmcore: unaligned memory access on RISCV64 targets

I-mikan-I, 2023-04-19:
## Description

ems_gc defines the macro `GC_HEAD_PADDING 4`, which causes the heap's base to be 32-bit aligned and trigger unaligned access faults on 64 bit targets.

## Replicate

Target used: kendryte k210
main.c:
```c
int main()
{
    char error_buf[128];
    wasm_module_t module;
    wasm_module_inst_t module_inst;
    wasm_function_inst_t func;
    wasm_exec_env_t exec_env;
    uint32_t size, stack_size = 8092, heap_size = 8092;

    /* initialize the wasm runtime by default configurations */
    RuntimeInitArgs args = {
        .mem_alloc_type = Alloc_With_System_Allocator,
    };
    wasm_runtime_full_init(&args);

    /* parse the WASM file from buffer and create a WASM module */
    module = wasm_runtime_load(aha_mont64, sizeof aha_mont64, error_buf, sizeof(error_buf));

    /* create an instance of the WASM module (WASM linear memory is ready) */
    module_inst = wasm_runtime_instantiate(module, stack_size, heap_size,
                                           error_buf, sizeof(error_buf));

    func = wasm_runtime_lookup_function(module_inst, "_start", NULL);
    exec_env = wasm_runtime_create_exec_env(module_inst, stack_size);
    wasm_val_t results[1];
    if (wasm_runtime_call_wasm_v(exec_env, func, 1, results, 0))
    {
        printf("wasm function return: %d\n", results[0].of.i32);
    }
    else
    {
        printf("error executing wasm function!\n%s\n", wasm_runtime_get_exception(module_inst));
    }

    wasm_runtime_destroy_exec_env(exec_env);
    wasm_runtime_deinstantiate(module_inst);
    wasm_runtime_unload(module);
    wasm_runtime_destroy();
}
```
CMakeLists.txt:
```cmake
set (SHARED_PLATFORM_CONFIG src/platform/shared_platform.cmake)
set (WAMR_BUILD_PLATFORM "platform")
set (WAMR_BUILD_TARGET "RISCV64_LP64")
set (WAMR_BUILD_INTERP 1)
set (WAMR_BUILD_FAST_INTERP 0)
set (WAMR_BUILD_LIBC_BUILTIN 1)
set (WAMR_DISABLE_HW_BOUND_CHECK 0)
set (WAMR_DISABLE_STACK_HW_BOUND_CHECK 0)
set (WAMR_ROOT_DIR wasm-micro-runtime)
```
platform_api_vmcore.c
```c
#include "platform_api_vmcore.h"
#include <stdio.h>
#include <stdarg.h>
/****************************************************
 *                     Section 1                    *
 *        Interfaces required by the runtime        *
 ****************************************************/

/**
 * Initialize the platform internal resources if needed,
 * this function is called by wasm_runtime_init() and
 * wasm_runtime_full_init()
 *
 * @return 0 if success
 */
int
bh_platform_init(void) {
    return 0;
}

/**
 * Destroy the platform internal resources if needed,
 * this function is called by wasm_runtime_destroy()
 */
void
bh_platform_destroy(void) {
    return 0;
}

/**
 ******** memory allocator APIs **********
 */

void *
os_malloc(unsigned size) {
    return malloc(size);
}

void *
os_realloc(void *ptr, unsigned size) {
    return realloc(ptr, size);
}

void
os_free(void *ptr) {
    return free(ptr);
}

/**
 * Note: the above APIs can simply return NULL if wasm runtime
 *       isn't initialized with Alloc_With_System_Allocator.
 *       Refer to wasm_runtime_full_init().
 */

int
os_printf(const char *format, ...) {
    va_list argptr;
    va_start(argptr, format);
    int result = vfprintf(stdout, format, argptr);
    va_end(argptr);
    return result;
}

int
os_vprintf(const char *format, va_list ap) {
    return vfprintf(stdout, format, ap);
}

/**
 * Get microseconds after boot.
 */
uint64
os_time_get_boot_microsecond(void) {
    return 0;
}

/**
 * Get current thread id.
 * Implementation optional: Used by runtime for logging only.
 */
korp_tid
os_self_thread(void) {
    return NULL;
}

extern uint64_t *__tp0;
/**
 * Get current thread's stack boundary address, used for runtime
 * to check the native stack overflow. Return NULL if it is not
 * easy to implement, but may have potential issue.
 */
uint8 *
os_thread_get_stack_boundary(void) {
    return (uint8 *)__tp0;
}

/**
 ************** mutext APIs ***********
 *  vmcore:  Not required until pthread is supported by runtime
 *  app-mgr: Must be implemented
 */

int
os_mutex_init(korp_mutex *mutex) {
    return 0;
}

int
os_mutex_destroy(korp_mutex *mutex) {
    return 0;
}

int
os_mutex_lock(korp_mutex *mutex) {
    return 0;
}

int
os_mutex_unlock(korp_mutex *mutex) {
    return 0;
}
```

I-mikan-I, 2023-04-19:
Note, the problematic code is in `ems_kfc.c`:

```c
gc_init_with_pool:
//...
base_addr =
        (char *)(((uintptr_t)base_addr + 7) & (uintptr_t)~7) + GC_HEAD_PADDING;
//...

gc_init_internal:
//...
q = (hmu_tree_node_t *)heap->base_addr;
q->size = heap->current_size; // STORE FAULT
```

wenyongh, 2023-04-20:
Hi, `q->size` is uint32 tyte and it should be not an issue to access it. My understanding is the accessing of `q->parent` causes the issue. I uploaded PR #2140 to fix it, could you help have a try?

I-mikan-I, 2023-04-20:
Doesn't seem to fix the issue:
```
V (18820889) SYSCALL: misaligned store recovered at 8001a990. len:08,addr:8004cd0c,reg:09,data:000000008004ac24,float:0                                                
core dump: misaligned load                                                                                                                                             
Cause 0x0000000000000004, EPC 0x0000000080007fbe                                                                                                                       
reg[00](zero ) = 0x0000000000000000, reg[01](ra   ) = 0x0000000080000222                                                                                               
reg[02](sp   ) = 0x00000000800311a0, reg[03](gp   ) = 0x00000000800236a0                                                                                               
reg[04](tp   ) = 0x00000000800298c0, reg[05](t0   ) = 0x0000000080000af8                                                                                               
reg[06](t1   ) = 0x0000000080007faa, reg[07](t2   ) = 0xffffffffffffffff                                                                                               
reg[08](s0/fp) = 0x000000008001a99a, reg[09](s1   ) = 0x000000008004ac24                                                                                               
reg[10](a0   ) = 0x0000000000000006, reg[11](a1   ) = 0x000000008001a99a                                                                                               
reg[12](a2   ) = 0x00000000800313c0, reg[13](a3   ) = 0x00000000800314c0                                                                                               
reg[14](a4   ) = 0x0000000000000003, reg[15](a5   ) = 0x0000000000000003                                                                                               
reg[16](a6   ) = 0x0000000080022070, reg[17](a7   ) = 0x00000000000000d6                                                                                               
reg[18](s2   ) = 0x000000008004ccf4, reg[19](s3   ) = 0x0000000000000001                                                                                               
reg[20](s4   ) = 0x000000008004cbc0, reg[21](s5   ) = 0x000000008004cbc0                                                                                               
reg[22](s6   ) = 0x0000000080031718, reg[23](s7   ) = 0x0000000000000080                                                                                               
reg[24](s8   ) = 0x0000000080039b20, reg[25](s9   ) = 0x0000000000000001                                                                                               
reg[26](s10  ) = 0x00000000800399e0, reg[27](s11  ) = 0x0000000000001fa0                                                                                               
reg[28](t3   ) = 0x000000000000001f, reg[29](t4   ) = 0x00000000000000f4                                                                                               
reg[30](t5   ) = 0x0000000080039bf0, reg[31](t6   ) = 0x00000000000000e0                                                                                               
freg[00](ft0 ) = 0x0000000000000000(), freg[00]() = 0x000000008001c278()                                                                                               
freg[02](ft2 ) = 0x0000000000000000(), freg[00]() = 0x000000008001c288()                                                                                               
freg[04](ft4 ) = 0x3432636100000000(), freg[-1073741824]() = 0x000000008001c298()                                                                                      
freg[06](ft6 ) = 0x0000000000000000(), freg[00]() = 0x000000008001c2a8()                                                                                               
freg[08](fs0 ) = 0x0000000000000000(), freg[00]() = 0x000000008001c2d0()                                                                                               
freg[10](fa0 ) = 0x0000000000000000(), freg[00]() = 0x000000008001c300()                                                                                               
freg[12](fa2 ) = 0x0000000000000000(), freg[00]() = 0x000000008001c328()                                                                                               
freg[14](fa4 ) = 0x0000000000000000(), freg[00]() = 0x000000008001c338()                                                                                               
freg[16](fa6 ) = 0x0000000000000000(), freg[00]() = 0x000000008001c348()                                                                                               
freg[18](fs2 ) = 0x0000000000000000(), freg[00]() = 0x000000008001c370()                                                                                               
freg[20](fs4 ) = 0x0000000000000000(), freg[00]() = 0x000000008001c380()                                                                                               
freg[22](fs6 ) = 0x0000000000000000(), freg[00]() = 0x000000008001c390()                                                                                               
freg[24](fs8 ) = 0x0000000000000000(), freg[00]() = 0x000000008001c3a0()                                                                                               
[... cut ...]
   0x000000008001a9ae <+208>:   13 35 95 0f     sltiu   a0,a0,249
   0x000000008001a9b2 <+212>:   97 50 ff ff     auipc   ra,0xffff5
   0x000000008001a9b6 <+216>:   e7 80 60 84     jalr    -1978(ra) # 0x8000f1f8 <bh_assert_internal>

47
48          return heap;
   0x000000008001a9ba <+220>:   8d b7   j       0x8001a91c <gc_init_internal+62>
End of assembler dump.
```

The issue might be that the structs aren't packed, so the pointers will always be aligned to 8-byte boundaries within the struct.

wenyongh, 2023-04-20:
The failure seems to be here:
```bash
42          root->right = q;
   0x000000008001a990 <+178>:   23 3e 94 12     sd      s1,316(s0)

43          q->parent = root;
```
Maybe you can add printf to dump the value of `root`, `&root->right` and `&q->parent`?

wenyongh, 2023-04-21:
@I-mikan-I I submitted a patch set to fix the issues, could you please try again?