wasm-micro-runtime/wasm-micro-runtime #2149: c_api_func_imports are not passed to new threads

cpetig, 2023-04-24:
While combining wasi-threads with custom functions defined by the executing program I found that on newly created threads the call fails with "Exception: failed to call unlinked import function (...)".

c_api_func_imports is not properly passed to new threads, copying the following code snippet from lib_pthread_wrapper.c into wasm_runtime.c fixes the problem for me:
[c_api_imports.patch](https://github.com/bytecodealliance/wasm-micro-runtime/files/11315198/c_api_imports.patch)

lum1n0us, 2023-04-25:
Would you mind sharing the test code or the specific scenario can trigger the issue?

cpetig, 2023-04-25:
I was able to reproduce the problem with wasm-c-api-imports. if you use sdk-20 and replace --target=wasm32-wasi with --target=wasm32-wasi-threads the old pthread_foo symbols get replaced by the standardized wasi.thread-spawn. 

Running this with example1 yields:
Extracting export...
Calling _start ...
[...]: .../wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_classic.c, line 4310, meet an exception Exception: failed to call unlinked import function (env, log)

cpetig, 2023-05-02:
I found that my patch is slightly wrong, the size_in_bytes should be sizeof(CApiFuncImport)*import_func_count (no star, thus size of struct not the pointer)

eloparco, 2023-05-03:
I managed to reproduce. Depending on the run, I get an unaligned exception
```
Initializing...
Loading binary...
> Load .wasm
Compiling module...
[17:04:39:988 - 209814140]: Load type section success.

[17:04:39:118 - 209814140]: Load import section success.

[17:04:39:573 - 209814140]: Load function section success.

[17:04:39:586 - 209814140]: Load table section success.

[17:04:39:591 - 209814140]: Load memory section success.

[17:04:39:595 - 209814140]: Load global section success.

[17:04:39:600 - 209814140]: Load export section success.

[17:04:39:602 - 209814140]: Load start section success.

[17:04:39:608 - 209814140]: Load table segment section success.

[17:04:39:611 - 209814140]: Load datacount section success.

[17:04:39:613 - 209814140]: Load code segment section success.

[17:04:39:620 - 209814140]: Load data segment section success.

[17:04:39:623 - 209814140]: Ignore custom section [name].
[17:04:39:625 - 209814140]: Ignore custom section [producers].
[17:04:39:627 - 209814140]: Ignore custom section [target_features].
[17:04:39:628 - 209814140]: Found aux __heap_base global, value: 39104
[17:04:39:630 - 209814140]: Found aux __data_end global, value: 6336
[17:04:39:632 - 209814140]: Found aux stack top global, value: 39104, global index: 0, stack size: 32768
[17:04:39:786 - 209814140]: /Users/eloparco/dev/forks/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c, line 715, can not find an export 0 named _initialize in the module 
[17:04:39:792 - 209814140]: Load module success.

Instantiating module...
[17:04:39:994 - 209814140]: Reset __heap_base global to 71872
[17:04:39:997 - 209814140]: Memory instantiate:
[17:04:39:999 - 209814140]:   page bytes: 65536, init pages: 2, max pages: 4
[17:04:39:001 - 209814140]:   heap offset: 39104, heap size: 32768

[17:04:39:043 - 209814140]: Memory instantiate success.
Extracting export...
Calling _start ...
[17:04:39:566 - 16D163000]: /Users/eloparco/dev/forks/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_classic.c, line 4310, meet an exception Exception: unaligned atomic
[17:04:39:575 - 16D177000]: /Users/eloparco/dev/forks/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_classic.c, line 4310, meet an exception Exception: unaligned atomic
[17:04:40:719 - 209814140]: /Users/eloparco/dev/forks/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_classic.c, line 4310, meet an exception Exception: unaligned atomic
[17:04:40:819 - 209814140]: /Users/eloparco/dev/forks/wasm-micro-runtime/core/iwasm/common/wasm_c_api.c, line 3326, Exception: unaligned atomic
> Error calling function! Exception: unaligned atomic
```

or (as you described) an import error
```
Initializing...
Loading binary...
> Load .wasm
Compiling module...
[17:05:33:273 - 209814140]: Load type section success.

[17:05:33:388 - 209814140]: Load import section success.

[17:05:33:869 - 209814140]: Load function section success.

[17:05:33:877 - 209814140]: Load table section success.

[17:05:33:882 - 209814140]: Load memory section success.

[17:05:33:887 - 209814140]: Load global section success.

[17:05:33:891 - 209814140]: Load export section success.

[17:05:33:894 - 209814140]: Load start section success.

[17:05:33:898 - 209814140]: Load table segment section success.

[17:05:33:901 - 209814140]: Load datacount section success.

[17:05:33:904 - 209814140]: Load code segment section success.

[17:05:33:910 - 209814140]: Load data segment section success.

[17:05:33:913 - 209814140]: Ignore custom section [name].
[17:05:33:914 - 209814140]: Ignore custom section [producers].
[17:05:33:916 - 209814140]: Ignore custom section [target_features].
[17:05:33:918 - 209814140]: Found aux __heap_base global, value: 39104
[17:05:33:921 - 209814140]: Found aux __data_end global, value: 6336
[17:05:33:923 - 209814140]: Found aux stack top global, value: 39104, global index: 0, stack size: 32768
[17:05:33:096 - 209814140]: /Users/eloparco/dev/forks/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c, line 715, can not find an export 0 named _initialize in the module 
[17:05:33:112 - 209814140]: Load module success.

Instantiating module...
[17:05:33:258 - 209814140]: Reset __heap_base global to 71872
[17:05:33:261 - 209814140]: Memory instantiate:
[17:05:33:263 - 209814140]:   page bytes: 65536, init pages: 2, max pages: 4
[17:05:33:265 - 209814140]:   heap offset: 39104, heap size: 32768

[17:05:33:308 - 209814140]: Memory instantiate success.
Extracting export...
Calling _start ...
[17:05:33:877 - 16B49B000]: /Users/eloparco/dev/forks/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_classic.c, line 4310, meet an exception Exception: failed to call unlinked import function (env, log)
[17:05:33:880 - 16B4AF000]: /Users/eloparco/dev/forks/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_classic.c, line 4310, meet an exception Exception: failed to call unlinked import function (env, log)
[17:05:34:918 - 209814140]: /Users/eloparco/dev/forks/wasm-micro-runtime/core/iwasm/interpreter/wasm_interp_classic.c, line 4310, meet an exception Exception: failed to call unlinked import function (env, log)
[17:05:34:966 - 209814140]: /Users/eloparco/dev/forks/wasm-micro-runtime/core/iwasm/common/wasm_c_api.c, line 3326, Exception: failed to call unlinked import function (env, log)
> Error calling function! Exception: failed to call unlinked import function (env, log)
```

I'm using wasi-sdk 20 and this [add-threads.patch](https://github.com/bytecodealliance/wasm-micro-runtime/files/11382709/add-threads.patch) on top of the `main` branch.

cpetig, 2023-05-03:
I added a testcase to https://github.com/bytecodealliance/wasm-micro-runtime/issues/2172 - this shows both issues, this one first

eloparco, 2023-05-04:
Thanks for the investigation and the changes. @cpetig where you getting the `unaligned atomic` error, too? If not, can you share the patch to use `wasm32-wasi-threads`? Maybe I'm doing something wrong there.

cpetig, 2023-05-04:
> where you getting the `unaligned atomic` error, too? If not, can you share the patch to use `wasm32-wasi-threads`? Maybe I'm doing something wrong there.

My testcase is https://github.com/bytecodealliance/wasm-micro-runtime/issues/2172#issuecomment-1533816513 
But I have never seen unaligned atomic - but I would suspect that the incomplete copying of the import table might have any effect.

eloparco, 2023-05-05:
> My testcase is https://github.com/bytecodealliance/wasm-micro-runtime/issues/2172#issuecomment-1533816513
But I have never seen unaligned atomic - but I would suspect that the incomplete copying of the import table might have any effect.

I see, the `unaligned atomic` error doesn't show up when I use your testcase, but it does with [`send_recv.c`](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/samples/wasm-c-api-imports/wasm/send_recv.c).
I tried it with the `main` branch, where the fix #2173 has already been merged.

wenyongh, 2023-05-10:
Close since the issue was resolved with #2173.

closed 2023-05-10 by wenyongh