wasm-micro-runtime/wasm-micro-runtime #2207: Propagate a SIGINT to the WASM module

dzobbe, 2023-05-13:
Hi All,

I have an infinite loop in a WASM module and I would like to setup a graceful exit. I discovered that WASI provides support for defining signal handler by defining this option `_WASI_EMULATED_SIGNAL` . 
However, it seems that WAMR does not propagate the ctrl+c signal and the handler is not called.
Is there some way to do it or is it just not available?

Thanks in advance.

wenyongh, 2023-05-15:
Hi, a thread will be notified and required to exit when there is an exception thrown or `wasi_exit` occurs, you may add check in your loop:
```C
#if WASM_ENABLE_THREAD_MGR != 0
#include "thread-mgr/thread_manager.h"
#endif

if (wasm_runtime_get_exception(module_inst)
    || wasm_cluster_is_thread_terminated(exec_env)) {
    //exit loop
}
```
Refer to
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/libraries/libc-wasi/libc_wasi_wrapper.c#L1036

yamt, 2023-05-15:
> Hi All,
>
> I have an infinite loop in a WASM module and I would like to setup a graceful exit. I discovered that WASI provides support for defining signal handler by defining this option [...]

it just isn't available in WASI.

`_WASI_EMULATED_SIGNAL` is just an emulation of `raise` within libc.

wenyongh, 2023-05-15:
> it just isn't available in WASI.
>
> `_WASI_EMULATED_SIGNAL` is just an emulation of `raise` within libc.

Thanks for the reminder, is there any document and sample?

yamt, 2023-05-15:
> Thanks for the reminder, is there any document and sample?

i'm not aware of anything other than the source code: https://github.com/WebAssembly/wasi-libc/blob/main/libc-bottom-half/signal/signal.c

dzobbe, 2023-05-15:
> Hi, a thread will be notified and required to exit when there is an exception thrown or `wasi_exit` occurs, you may add check in your loop:
>
> ```c
> #if WASM_ENABLE_THREAD_MGR != 0
> #include [...]

Thanks. But this is executable outside of the wasm module right? I would like to intercept the termination in the module itself.