wasm-micro-runtime/wasm-micro-runtime #2220: thread-manager: cannot spawn thread without aux stack

JamesDunne, 2023-05-16:
Trying to use wamr's thread-manager API to spawn a dedicated thread to execute a simple wasi module which itself spawns no threads. The problem is that current [gotip](https://pkg.go.dev/golang.org/dl/gotip)`wasip1` feature has no support for specifying an "aux stack" as far as I can tell. This causes thread-manager to fail to spawn the thread due to no aux stack available.

My host app needs to load multiple wasi modules simultaneously with each module independent of the others; no memory/resource sharing. The host app spawns a dedicated thread to run each wasi module in. A key requirement for my app is to be able to dynamically unload a wasi module and cancel any running wasm code arbitrarily. I also would like to use Go to code the wasi modules.

JamesDunne, 2023-05-17:
I was able to get this scenario to work by simply enabling the thread-manager feature but managing the thread outside of thread-manager. This brings in the `wasm_cluster_cancel_thread()` which signals the interpreter to stop which is exactly what I need.

It might be a good idea to separate cancellation/interruption of wasm code from thread management.

closed 2023-05-17 by JamesDunne

mkokryashkin, 2024-04-23:
I would appreciate it if you could explain your solution in more detail. I've encountered the same issue and can't get around it.