wasm-micro-runtime/wasm-micro-runtime #2234: Zephyr ESP32 custom linker scripts not working properly

nightf0rc3, 2023-05-24:
Hi, I'm currently trying to build the Zephyr sample with an ESP32 target. I followed the instructions [here](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/product-mini/README.md#zephyr) and building for an `x86` target worked like a charm. When changing to `esp32` I was getting some errors at the linking step:
```
/root/wasm-micro-runtime/product-mini/platforms/zephyr/simple/esp32_custom_linker.ld:14:10: fatal error: devicetree.h: No such file or directory
   14 | #include <devicetree.h>
``` 
Which I fixed by prefixing all the zephyr imports in `esp32_custom_linker.ld` with `zephyr/`. After that I received the following error on the next build:
```
[219/229] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /root/wasm-micro-runtime/product-mini/platforms/zephyr/simple/build/zephyr/zephyr_pre0.map 
: && ccache /root/zephyr-sdk-0.16.1/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/root/wasm-micro-runtime/product-mini/platforms/zephyr/simple/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/root/zephyr-sdk-0.16.1/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0"  -L/root/wasm-micro-runtime/product-mini/platforms/zephyr/simple/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -lgcc  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -T/root/zephyrproject/modules/hal/espressif/zephyr/esp32/src/linker/esp32.rom.alias.ld  -T/root/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.ld  -T/root/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.api.ld  -T/root/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.libgcc.ld  -T/root/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/soc/esp32/ld/esp32.peripherals.ld  -Wl,--wrap=k_calloc  -Wl,--wrap=k_malloc && cd /root/wasm-micro-runtime/product-mini/platforms/zephyr/simple/build/zephyr && /usr/bin/cmake -E echo
/root/zephyr-sdk-0.16.1/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd:zephyr/linker_zephyr_pre0.cmd:115: syntax error
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /root/wasm-micro-runtime/product-mini/platforms/zephyr/simple/build
```
Is there any step I'm missing or is there something wrong with the custom linker script?

Thanks in advance!

### Hardware & OS
Ubuntu 22.04
CPU: shared AMD EPYC™ 7281 (2.1 GHz)
Memory: 4GB

TianlongLiang, 2023-06-09:
Hi, I will try to have the Zephyr environment installed and reproduce this issue ASAP, it could be a while(I sure hope it won't be too long) because I am a rookie in Zephyr.

TianlongLiang, 2023-06-12:
I have reproduced that issue, I guess this is because the linker scripts `esp32_custom_linker.ld` haven't updated for corresponding changes in Zephyr. I will try to fix it, but if you want to use it immediately, you can try a lower version of Zephyr and Espressif

nightf0rc3, 2023-06-13:
Thanks for the feedback and your work! I'll try a lower version when I find some time.

TianlongLiang, 2023-06-14:
I am also trying to find a lower working version, and considering adding a Dockerfile with that fixed version. If you beat me to find that version first, I would appreciate it if you could share it with me :)

TianlongLiang, 2023-06-26:
In [this pr](https://github.com/bytecodealliance/wasm-micro-runtime/pull/2314), the Dockerfile.old should be able to provide a lower version of Zephyr that works with ESP32 custom linker script. If you want to install an older version of Zephyr locally, you can try 2.3.0 with SDK 0.11.3, and esp-idf 4.0

TianlongLiang, 2023-06-28:
@nightf0rc3 Let me know whether this dockerfile works for you or let know if there is any problem