wasm-micro-runtime/wasm-micro-runtime #2235: Run AOT failed on windows

Waitforomething2023, 2023-05-24:
Os:windows 10
wamrc_ver:latest x86
toolchain:emcc
llvm_ver:15.X

```cpp
#include <iostream>

int main()
{

    std::cout<<"Hello, world!\n";
    return 0;

}

```

cmooand:wamrc --target-abi=msvc --target=i386 -o test.aot test.wasm

```log
Create AoT compiler with:
  target:        i386
  target cpu:
  target triple: i386-pc-windows-msvc
  cpu features:
  opt level:     3
  size level:    1
  output format: AoT file
Compile success, file test.aot was generated.

iwasm.exe test.aot
Load relocation section failed: invalid relocation type 20.

wenyongh, 2023-05-25:
Hi, the relocations for Windows x86-32 haven't been implemented yet, I tried to enable the relocation type 20 as you mentioned, see [win_x86_32.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/11561566/win_x86_32.zip), could you have a try?

Waitforomething2023, 2023-05-25:
> Hi, the relocations for Windows x86-32 haven't been implemented yet, I tried to enable the relocation type 20 as you mentioned, see [win_x86_32.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/11561566/win_x86_32.zip), could you have a try?

Hi I've tried,buf it failed.
```log

AOT module load failed: resolve symbol _aot_func#384 failed
```

wenyongh, 2023-05-26:
It seems that the function symbol generated in Windows 32bit starts with '_', could you modify the aot_loader.c:
```diff
diff --git a/core/iwasm/aot/aot_loader.c b/core/iwasm/aot/aot_loader.c
index 5345fb2d..9a431d02 100644
--- a/core/iwasm/aot/aot_loader.c
+++ b/core/iwasm/aot/aot_loader.c
@@ -1947,6 +1947,19 @@ do_text_relocation(AOTModule *module, AOTRelocationGroup *group,
             }
             symbol_addr = module->func_ptrs[func_index];
         }
+#if defined(BH_PLATFORM_WINDOWS) && defined(BUILD_TARGET_X86_32)
+        else if (!strncmp(symbol, "_" AOT_FUNC_PREFIX,
+                          strlen("_" AOT_FUNC_PREFIX))) {
+            p = symbol + strlen("_" AOT_FUNC_PREFIX);
+            if (*p == '\0'
+                || (func_index = (uint32)atoi(p)) > module->func_count) {
+                set_error_buf_v(error_buf, error_buf_size, "invalid symbol %s",
+                                symbol);
+                goto check_symbol_fail;
+            }
+            symbol_addr = module->func_ptrs[func_index];
+        }
+#endif
         else if (!strcmp(symbol, ".text")) {
             symbol_addr = module->code;
         }
```

Waitforomething2023, 2023-05-26:
> It seems that the function symbol generated in Windows 32bit starts with '_', could you modify the aot_loader.c:
>
> ```diff
> diff --git a/core/iwasm/aot/aot_loader.c b/core/iwasm/aot/aot_loader.c
> index [...]

worked.

Waitforomething2023, 2023-05-26:
> Seems that `__aulldiv` is a compiler-rt function, could you try declaring it in `aot_reloc_x86_32.c` and then `REG_SYM` it:
> 
> https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/aot/arch/aot_reloc_x86_32.c#L52-L55
> 
> ```c
> void __aulldiv();
> 
> static SymbolMap target_sym_map[] = {
>     ...
>     REG_SYM(__aulldiv),
> };
> ```

ok. 

```cpp
static uint64
__aulldiv(uint64 a, uint64 b)
{
    return a / b;
}
#endif

/* clang-format off */
static SymbolMap target_sym_map[] = {
    REG_COMMON_SYMBOLS
    /* compiler-rt symbols that come from compiler(e.g. gcc) */
    REG_SYM(__divdi3),
    REG_SYM(__udivdi3),
    REG_SYM(__moddi3),
    REG_SYM(__umoddi3),
    REG_SYM(__aulldiv)
};
```

runlog:

```log
Load relocation section failed: invalid relocation type 6.
```

wenyongh, 2023-05-27:
In aot_reloc_x86_32.c, try adding `#define IMAGE_REL_I386_REL32 6` and then:
```diff
     switch (reloc_type) {
+#if !defined(BH_PLATFORM_WINDOWS)
         case R_386_32:
+#else
+        case IMAGE_REL_I386_DIR32:
+#endif
```

Waitforomething2023, 2023-05-28:
> ```diff
> case IMAGE_REL_I386_DIR32:
> ```

tried,then:
```log
Load relocation section failed: invalid relocation type 20.
```

wenyongh, 2023-05-29:
Hi, refer to below link for Windows x86-32 relocations:
https://learn.microsoft.com/en-us/windows/win32/debug/pe-format#intel-386-processors

It seems the implementation of the two relocation types should be like:
IMAGE_REL_I386_REL32  (=20):  refer to the patch [win_x86_32.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/11561566/win_x86_32.zip)
IMAGE_REL_I386_DIR32 (=6): refer to the patch:
```diff
     switch (reloc_type) {
+#if !defined(BH_PLATFORM_WINDOWS)
         case R_386_32:
+#else
+        case IMAGE_REL_I386_DIR32:
+#endif
```

wenyongh, 2023-06-12:
@Waitforomething2023 Thanks for reporting the issues, I uploaded PR #2285 to fix them, could you help check whether there is anything missing? Thanks.