wasm-micro-runtime/wasm-micro-runtime #2242: aot_loader RtlAddFunctionTable logic seems to assume a particular order of functions

yamt, 2023-05-29:
the aot_loader logic to build a table for RtlAddFunctionTable seems to have an assumption that functions are ordered by their code addresses.
https://github.com/bytecodealliance/wasm-micro-runtime/blob/76be848ec34ebd3285fa5840a1db1e81c9e0f941/core/iwasm/aot/aot_loader.c#L1675-L1681
however, as far as understand, LLVMAddFunction order isn't guaranteed to be preserved in the object file.
(LLVM is free to change the order for optimizations, pc-relative addressing, etc.)
and i couldn't find anything is our aot compiler which ensures the expected order.

yamt, 2023-06-08:
as we specifiy the same UnwindInfoAddress for all entries, i wonder if it's ok to use a single entry covering the whole code section instead.

yamt, 2023-06-13:
> as we specify the same UnwindInfoAddress for all entries, i wonder if it's ok to use a single entry covering the whole code section instead.

@wenyongh any idea?

wenyongh, 2023-06-13:
@yamt Hi, yes, currently the implementation of AOT hw bound check for windows 64-bit has defects:

(1) Using RtlAddFunctionTable to dynamically register the AOT functions to the system is the ideal way if we can unwind the AOT functions in the calling stack, and the exception of accessing guard pages can be caught in the aot_runtime.c `invoke_native_with_hw_bound_check` by using `__try { ... } __except(..) { .. }` sentence:
  https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/aot/aot_runtime.c#L1387-L1411

Note that in the interpreter mode, we use `__try { ... } __except(..) { .. }` and it works:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/interpreter/wasm_runtime.c#L2372-L2379

But here I am not sure how to make the AOT functions be unwound successfully, the exception can be caught in `runtime_exception_handler` (registered by AddVectoredExceptionHandler):
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/common/wasm_runtime_common.c#L204

while it cannot be forwarded to the `except` sentence of `__try { .. } __except(..) { .. }` when we return EXCEPTION_CONTINUE_SEARCH in runtime_exception_handler. So I changed the `RIP` and let aot code continue to execute:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/common/wasm_runtime_common.c#L242-L248
```C
    exce_info->ContextRecord->Rip++;
    return EXCEPTION_CONTINUE_EXECUTION;
```
So in fact the RtlAddFunctionTable doesn't take effect currently.

(2) But the mode of `EXCEPTION_CONTINUE_EXECUTION` is also not a good way, recently I found it also has issues: (a) Increasing Rip with one byte may point to an invalid instruction, we need to decode the instruction and get its size to let `Rip += size` to point to the next instruction and continue to run, (b) letting the aot code continue run may cause unexpected behavior, we had better let it return/exit as soon as possible.

For your question, (1) yes, the code pieces of aot functions may be unordered, normally they are ordered, but I found that after generating AOT file with PGO profile data, they can be unordered, (2) Registering the AOT code with the whole code section seems not a good way, not sure whether it will cause unexpected behavior when unwinding, for example, the expected call stack is `inovkeNative->aot_func#3->aot_func#1->aot_func#2`, but after that, it becomes `invokeNative->unknown position` suppose that `aot_func#3` is not the beginning of the code section.

I think maybe we can sort the aot function addresses firstly to make them ordered, and register the ordered functions, e.g. [aot_func#2, aot_func#3], [aot_func#3, aot_func#1], [aot_func#1, code_end].

yamt, 2023-06-13:
> @yamt Hi, yes, currently the implementation of AOT hw bound check for windows 64-bit has defects:
>
> (1) Using RtlAddFunctionTable to dynamically register the AOT functions to the system is the ideal [...]

does it work for JIT?
compiler-rt seems to have RtlAddFunctionTable stuff for orc ExecutionEngine.

> So I changed the `RIP` and let aot code continue to execute: https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/common/wasm_runtime_common.c#L242-L248
> 
> ```c
>     exce_info->ContextRecord->Rip++;
>     return EXCEPTION_CONTINUE_EXECUTION;
> ```

does it work at all?
a single-byte instruction is rather rare on x86-64.

> So in fact the RtlAddFunctionTable doesn't take effect currently.
>
> (2) But the mode of `EXCEPTION_CONTINUE_EXECUTION` is also not a good way, recently I found it also has issues: (a) Increasing Rip [...]

isn't it something we can make LLVM generate?
does our aot file contains a copy of `.pdata` section?

wenyongh, 2023-06-14:
> 
> does it work for JIT? compiler-rt seems to have RtlAddFunctionTable stuff for orc ExecutionEngine.
> 

No, I have a test, the exception cannot be caught also.

> > ```c
> >     exce_info->ContextRecord->Rip++;
> >     return EXCEPTION_CONTINUE_EXECUTION;
> > ```
> 
> does it work at all? a single-byte instruction is rather rare on x86-64.

It works for some cases.

> isn't it something we can make LLVM generate? does our aot file contains a copy of `.pdata` section?

Currently the text offsets stored in the AOT file are the offsets of each aot function, the function index isn't stored. Do you mean to store the ordered text offsets? If yes, we may need to store the function index also. The `.pdata` section isn't stored, what is that used for?

yamt, 2023-06-14:
> No, I have a test, the exception cannot be caught also.
>
>
> It works for some cases.
>
>
> Currently the text offsets stored in the AOT file are the offsets of each aot function, the function index isn't [...]

my understanding is that for a normal executable, llvm generates `.pdata` section, which contains the table we want.
https://learn.microsoft.com/en-us/windows/win32/debug/pe-format#the-pdata-section

wenyongh, 2023-06-14:
Got it, I will try to test it. It includes `Unwind Information` in each function entry, not sure how it is stored, need more investigation.