wasm-micro-runtime/wasm-micro-runtime #2247: A question about the compilation determinism of `wamrc`。

ghost-787, 2023-06-01:
I am wondering if the compilation result is always the same for a given target platform and compilation options when compiling a WASM application. Specifically, if I use the same version (commit ID) of wamrc, will the resulting AoT file be the same when compiling a WASM file on an x86 machine running Ubuntu 20.04 compared to a RISC-V machine running Ubuntu 20.04? Additionally, does the operating system used for compilation have any influence on the result? Could you provide some examples of factors that may influence the compilation result?

Thank you for your time and expertise in advance.

TianlongLiang, 2023-06-05:
> If I use the same version (commit ID) of wamrc, will the resulting AoT file be the same when compiling a WASM file on an x86 machine running Ubuntu 20.04 compared to a RISC-V machine running Ubuntu 20.04?

Usually, it wouldn't be the same, including the example you use.

> Additionally, does the operating system used for compilation have any influence on the result?

Yes, OS will potentially influence the result.

> Could you provide some examples of factors that may influence the compilation result?

As far as I know, the factors that will affect the compiled AOT file content are:
- Target Architecture
- Target CPU
- Target ABI(Influenced by OS, because OS typically defines or chooses the ABI)

You can use the CLI argument to set those targets for `wamrc`, for example:
```shell
./wamrc --target=xxx --target-abi=xxx --cpu=xxx -o main.aot main.wasm
```

ghost-787, 2023-06-05:
Thanks for your kind answers!
I am sorry that I did not describe my problem clearly.
My question is that given the same *target* information (e.g. target architecture, target CPU, target ABI), will the architecture or the OS of the machine that perform the compilation, have an influent on the result AoT binary?
e.g. compile a hello.wasm to x86-AoT, on an x86 Ubuntu machine and on a RISC-V Ubuntu machine respectively.
Could you provide some examples of factors that may influence the compilation result, except the factors about the target platform?
Really appreciate your help!

yamt, 2023-06-06:
> My question is that given the same target information (e.g. target architecture, target CPU, target ABI)

be aware that there are subtle differences like: https://github.com/bytecodealliance/wasm-micro-runtime/blob/6e3c3fe9ec08faf0f28db91fd952d6bbe2334935/wamr-compiler/main.c#L390-L396

if you specifiy everything explicitly, hopefully it can produce the same (or at least compatible enough) binary. (unless you hit bugs like https://github.com/bytecodealliance/wasm-micro-runtime/pull/2260)

TianlongLiang, 2023-06-06:
> My question is that given the same target information (e.g. target architecture, target CPU, target ABI), will the architecture or the OS of the machine that perform the compilation, have an influent on the result AoT binary?

I can't be 100% sure, but theoretically speaking, it should be. 

The simplified workflow of how a aot file is generated:
llvm unopt ir -> llvm opt -> object file ->aot file

Let's take a closer look::

1. The input WebAssembly (wasm) file is first processed by wamrc, which generates unoptimized raw LLVM Intermediate Representation (IR).
2. This raw LLVM IR is then passed through the LLVM pass manager, where it is transformed into optimized LLVM IR.
3. From this optimized LLVM IR, an object file is created.
4. Finally, the object file, along with some extra data, is put into an AOT file.

You can dump the intermediate result to the file using those commands:

```shell
# it will dump raw llvm it
./wamrc --format=llvmir-unopt -o test_unopt.ll test.wasm
# it will dump optimized llvm it
./wamrc --format=llvmir-opt -o test_opt.ll test.wasm
# it will dump object file
./wamrc --format=object -o test_unopt.o test.wasm
# and of course, generate the aot file
./wamrc -o test.aot test.wasm
```
You can try those commands out with some sample programs. To see whether the result aligns with the supposed behavior (and if not, you can know which step it starts to differ)

closed 2023-06-09 by ghost-787