wasm-micro-runtime/wasm-micro-runtime #2248: enable TSAN for nightly build in WAMR

Zzzabiyaka, 2023-06-01:
For now if we try to run wamr tests with tsan it fails due to `signal-unsafe call inside of a signal`. 

We need to investigate this behaviour and fix the errors in order to enable tsan

yamt, 2023-07-11:
> For now if we try to run wamr tests with tsan it fails due to signal-unsafe call inside of a signal.

what errors does it produce?

glancing at the posix version of the code, at least it seems using a few unsafe functions:

in os_thread_get_stack_boundary
* pthread functions

in wasm_set_exception
* mutex lock/unlock
* snprintf

i'm thinking to fix them as:
* use a cached value instead of calling os_thread_get_stack_boundary
* do not call wasm_set_exception in the signal handler. do it after longjmp instead.

how do you think?

Zzzabiyaka, 2023-07-12:
Hi, 
Yeah, LGTM. 

Will try to take a couple of steps to implementation