wasm-micro-runtime/wasm-micro-runtime #2252: [BUG] GS register shouldn't be modified in linux-sgx platform hardware mode (64 bit)

LeoneChen, 2023-06-03:
Hello~

`get_thread_data` use `READ_TD_DATA` assembly macro [here](https://github.com/intel/linux-sgx/blob/1efe23c20e37f868498f8287921eedfbcecdc216/sdk/trts/linux/trts_pic.S#L78)
[image]

And in SGX hardware mode, `READ_TD_DATA` use `GS` register in 64bit mode as base address of thread data. [here](https://github.com/intel/linux-sgx/blob/1efe23c20e37f868498f8287921eedfbcecdc216/sdk/trts/linux/trts_pic.h#L106)
(Note in hardware mode `SE_SIM` is undefined)
[image]

Thus it shouldn't modify `GS` register in Enclave.[here for example](https://github.com/bytecodealliance/wasm-micro-runtime/blob/f1e9029ebcca71c357bfabfc0376c73c3b1d1d0d/core/iwasm/interpreter/wasm_interp_fast.c#L3987)
[image]

`os_writegsbase` macro is at [here](https://github.com/bytecodealliance/wasm-micro-runtime/blob/f1e9029ebcca71c357bfabfc0376c73c3b1d1d0d/core/shared/platform/linux-sgx/platform_internal.h#L63)

[image]

wenyongh, 2023-06-05:
@LeoneChen Thanks for pointing out this! I only tested in simulator mode and thought that GS register is available. It is a pity that we cannot use the register to enable the segue optimization for linux-sgx platform, I submitted PR #2255 to fix the issue.

LeoneChen, 2023-06-05:
My pleasure~

LeoneChen, 2023-06-05:
Hello~ We find another bug
# Heap OOB
It's about WAMR run in Enclave

In [EDL](https://github.com/bytecodealliance/wasm-micro-runtime/blob/5fb5119239220b0803e7045ca49b0a29fe65e70e/product-mini/platforms/linux-sgx/enclave-sample/Enclave/Enclave.edl#L24), size of `cmd_buf` is specified by `cmd_buf_size`, but `cmd_buf_size` is set by untrusted host.
[image]

When specify `[size=cmd_buf_size]` attribute, TBridge(*_t.c from *.edl) like below will malloc only `cmd_buf_size` for `cmd`.
```
static sgx_status_t SGX_CDECL sgx_ecall_handle_command(void* pms)
{
	...
	uint8_t* _tmp_cmd_buf = ms->ms_cmd_buf;
	unsigned int _tmp_cmd_buf_size = ms->ms_cmd_buf_size;
	size_t _len_cmd_buf = _tmp_cmd_buf_size;
	...

	if (_tmp_cmd_buf != NULL && _len_cmd_buf != 0) {
		if ( _len_cmd_buf % sizeof(*_tmp_cmd_buf) != 0)
		{
			status = SGX_ERROR_INVALID_PARAMETER;
			goto err;
		}
		_in_cmd_buf = (uint8_t*)malloc(_len_cmd_buf);
		if (_in_cmd_buf == NULL) {
			status = SGX_ERROR_OUT_OF_MEMORY;
			goto err;
		}

		if (memcpy_s(_in_cmd_buf, _len_cmd_buf, _tmp_cmd_buf, _len_cmd_buf)) {
			status = SGX_ERROR_UNEXPECTED;
			goto err;
		}
	}

	ecall_handle_command(ms->ms_cmd, _in_cmd_buf, _tmp_cmd_buf_size);
	...
}
```
Thus, in [ecall_handle_command](https://github.com/bytecodealliance/wasm-micro-runtime/blob/5fb5119239220b0803e7045ca49b0a29fe65e70e/product-mini/platforms/linux-sgx/enclave-sample/Enclave/Enclave.cpp#L605), `argc` can be any number (e.g. 0), and `cmd_buf` may only have insufficient heap memory space
[image]

However, e.g. in [handle_cmd_init_runtime](https://github.com/bytecodealliance/wasm-micro-runtime/blob/5fb5119239220b0803e7045ca49b0a29fe65e70e/product-mini/platforms/linux-sgx/enclave-sample/Enclave/Enclave.cpp#L97), this assert is only enabled in DEBUG mode, but this assert (or if-check) is necessary in RELEASE mode too.
[image]

When `assert` disabled in RELEASE mode, `args[0]` will cause HEAP OOB, which overflow heap memory prepared by SGXSDK.

wenyongh, 2023-06-05:
Hi, the `cmd_buf_size` of ecall_handle_command is specified when calling it from host for each command, like:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/product-mini/platforms/linux-sgx/enclave-sample/App/App.cpp#LL380C55-L380C55
https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/product-mini/platforms/linux-sgx/enclave-sample/App/App.cpp#L412

Not sure why "argc can be any number (e.g. 0), and cmd_buf may only have insufficient heap memory space", is there any case to reproduce the issue? Or do you mean that `cmd_buf_size` may be modified illegally in the host side?

LeoneChen, 2023-06-05:
Yes, I mean "cmd_buf_size may be modified illegally in the host side", since host may an attacker

LeoneChen, 2023-06-05:
I'm happy to help you fix SGX specific BUGs like this, if you need.

LeoneChen, 2023-06-05:
# Lack of check of Null-Pointer
`enclave_module` is passed by untrusted host, and can be NULL pointer, better to check `enclave_module` before use (e.g. `enclave_module->module`) ([here](https://github.com/bytecodealliance/wasm-micro-runtime/blob/5fb5119239220b0803e7045ca49b0a29fe65e70e/product-mini/platforms/linux-sgx/enclave-sample/Enclave/Enclave.cpp#L354))
[image]

LeoneChen, 2023-06-05:
>Yes, appreciate!
And note that these commands (init_runtime/load_module/instantiate_module, etc) which invoke ecall_handle_cmd are also called by wamr_pal_create_process. wamr_pal_create_process is called by the inclavare container project, we should ensure that the modification works for the project.
https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/product-mini/platforms/linux-sgx/enclave-sample/App
https://github.com/inclavare-containers/inclavare-containers

OK, I'll try to fix it

LeoneChen, 2023-06-05:
Hi~ I'll list all bugs found in this issue, and help to try to fix them~

# Time of Check to Time of Use/Double Fetch
In [wasm_runtime_load](https://github.com/bytecodealliance/wasm-micro-runtime/blob/5fb5119239220b0803e7045ca49b0a29fe65e70e/core/iwasm/common/wasm_runtime_common.c#L1116) (e.g. called by [ecall_iwasm_main](https://github.com/bytecodealliance/wasm-micro-runtime/blob/5fb5119239220b0803e7045ca49b0a29fe65e70e/product-mini/platforms/linux-sgx/enclave-sample/Enclave/Enclave.cpp#L685) ), `buf` is in host (non Enclave) memory, thus can be controlled by attacker.

[image]

[get_package_type](https://github.com/bytecodealliance/wasm-micro-runtime/blob/5fb5119239220b0803e7045ca49b0a29fe65e70e/core/iwasm/common/wasm_runtime_common.c#L638) will check first four bytes in `buf`, then `wasm_load` or `aot_load_from_aot_file` will try to load this `buf`.

But there is a situation that, attack at OS level can replace content of `buf` (reside in untrusted memory) just after the check of `get_package_type` but before loading of `wasm_load` or `aot_load_from_aot_file`. Thus when loading `buf`, the first four bytes can already be changed to anything.

Thus it's better to copy whole buffer into Enclave, and then check and use.

This problem is very similar to TOCTOU in linux kernel or other situations which involve security boundary. And is firstly pointed out by CCS'19 paper [A Tale of Two Worlds: Assessing the Vulnerability of Enclave Shielding Runtimes](https://jovanbulck.github.io/files/ccs19-tale.pdf). Or refer to my [Notes in Chinese language => Double Fetch](https://blog.csdn.net/clh14281055/article/details/117674987) if you want

wenyongh, 2023-06-06:
> Hi~ I'll list all bugs found in this issue, and help to try to fix them~
> 
Yes, many thanks!

LeoneChen, 2023-06-13:
# Null Pointer Dereference
Need to check `func_name` not Null before `strlen(func_name)`
https://github.com/bytecodealliance/wasm-micro-runtime/blob/5fb5119239220b0803e7045ca49b0a29fe65e70e/product-mini/platforms/linux-sgx/enclave-sample/Enclave/Enclave.cpp#L434

LeoneChen, 2023-06-30:
@wenyongh  Hello, the current interface designed for SGX is too open, but it's due to the requirement of fully supported web assembly runtime, and it seems too hard to fix. How about trying Gramine from Intel or Occlum from Ant Group to seamlessly support non-SGX version WAMR, they are LibOS solutions run in Enclave to support legacy applications and have well-designed Enclave interfaces

wenyongh, 2023-07-03:
@LeoneChen Yes, it may be very hard to fix, could you fix some simple cases? For Gramine and LibOS, maybe we can try running WAMR based on the platform linux (product-mini/platforms/linux), it should be OK if they supports the libc APIs that platform linux requires.

LeoneChen, 2023-07-03:
Hello!

My simple idea is change each handle in switch-case to ecalls, in order to leverage auto generated Tbridge code to memory clone for one-level pointer, but for nested pointer, we may need extra work.

This will change interface of Enclave, then host need change code to call these interface.

If you think it's ok, I will submit a PR for every to review

wenyongh, 2023-07-04:
It is good to me, but not sure whether it is OK for other developers, they might be using the current mode in their products. How about adding a macro to control it? By default the new interface is used, and developer can use `make WAMR_BUILD_XXX=1 (or 0)` to use the old interfaces, and in the App code, we can use macro `WASM_ENABLE_XXX=1/0` to control the code.

LeoneChen, 2023-07-04:
Ok

LeoneChen, 2023-07-13:
# UAF
## Malloc
`ecall_iwasm_main->wasm_runtime_full_init->wasm_runtime_env_init->wasm_shared_memory_init->bh_hash_map_create->wasm_runtime_malloc`
which set global `wait_map` 
https://github.com/bytecodealliance/wasm-micro-runtime/blob/81f0371f63ecfd1c3a91e13b086c546f5c67ebd2/core/iwasm/common/wasm_shared_memory.c#L49-L55

## Free
`ecall_iwasm_main->wasm_runtime_destroy->wasm_shared_memory_destroy->bh_hash_map_destroy->wasm_runtime_free`
which not set `wait_map` to NULL after free
https://github.com/bytecodealliance/wasm-micro-runtime/blob/81f0371f63ecfd1c3a91e13b086c546f5c67ebd2/core/iwasm/common/wasm_shared_memory.c#L66-L70

## Use
`ecall_handle_command->handle_cmd_destroy_runtime->wasm_runtime_destroy->wasm_shared_memory_destroy->bh_hash_map_destroy`
`map` is the `wait_map` passed from parameter and `map->lock` cause UAF
https://github.com/bytecodealliance/wasm-micro-runtime/blob/81f0371f63ecfd1c3a91e13b086c546f5c67ebd2/core/shared/utils/bh_hashmap.c#L249-L259
## Conclusion
WAMR forget to set `wait_map` to NULL after free, and ecall can be called arbitrarily by untrusted host.

LeoneChen, 2023-07-13:
Patch to fix Enclave interfaces is done, and I use fuzz to find potential bugs, after fuzz, I will pull request.

I'm a little busy recently, so I'm slow to fix above problem

wenyongh, 2023-07-18:
> PR #2357
> 
> I'm a little busy recently, so late to fix the above problems
> 
> One detail is that I use indexes to replace the raw pointers when passing `enclave_module` and `wasm_module_inst` across the SGX boundary. This method is decribed in [TeeRex](https://www.usenix.org/system/files/sec20-cloosters.pdf), and at the end of section 5.3
> 
> > The developers of the Rust SGX SDK acknowledged the problem and promptly updated their code. Akin to our suggestions to the developers, the enclave code now utilizes session identifiers instead of pointers to identify TLS sessions; similar to using file descriptors on Unix-like systems. Upon session creation in tls_client_new, the pointer to the TLS session object is now inserted into a hashmap, which is then used to map the identifier in subsequent ECALLs. Hence, no pointers are passed on the host-to-enclave boundary. This drastically reduces the attack surface of the enclave and eradicates both the vulnerability pattern Returning pointers to enclave memory (P2) and Pointers to Overlapping Memory (P3).
> 
> Since I'm not so familiar with WAMR, you may need further check. The Action on this PR failed on SPEC, but I have no idea about this.
> 
> I've fuzzed with this Enclave with the new interface, and no above bugs are reported anymore.

Thanks @LeoneChen! We will check the PR and try to fix the issues reported by GitHub CIs.