wasm-micro-runtime/wasm-micro-runtime #2257: Cannot build iwasm for JIT with external LLVM

matsbror, 2023-06-05:
I am having great issues building iwasm with JIT using my 16.0.5 release of llvm installed on my ubuntu 22.04.
I built LLVM with clang,lld and copmiler-rt and installed with `sudo make install`

This is my cmake command:

```
cmake .. -DWAMR_BUILD_INTERP=1 -DWAMR_BUILD_JIT=1  -DLLVM_DIR=/usr/local/lib/cmake/llvm
```

Configuration goes well and LLVM 16.0.5 is detected.

The first error I get is:
```
/usr/local/include/llvm/ADT/bit.h:64:56: error: ‘is_integral_v’ is not a member of ‘std’; did you mean ‘is_integral’?
```

This particular error I can get rid of by changing the C++ standard from 14 to 17 in the CMakeFile.txt.
However, then this error appears:

```
aot_llvm_extra.cpp:236:5: error: ‘Optional’ was not declared in this scope; did you mean ‘llvm::cl::Optional’?
  236 |     Optional<PGOOptions> PGO = None;
      |     ^~~~~~~~
      |     llvm::cl::Optional
```

What is the best way of building iwasm with an externally built LLVM?

matsbror, 2023-06-05:
Update: I can build it against branch `release/15.x` but not agains the latest version of llvm.

TianlongLiang, 2023-06-05:
Hi, so far, WAMR only supports LLVM 15.x, and earlier, I see you are able to use external LLVM with 15.x version. Will that version be enough for you to use?

matsbror, 2023-06-05:
Ok, that explains it.

closed 2023-06-05 by matsbror