wasm-micro-runtime/wasm-micro-runtime #2263: Linking issue with spawn thread API for ESP-IDF

skrovk, 2023-06-05:
### Description
Trying to build the (slightly modified for ESP) spawn thread example fails to build due to linking error - the spawn thread API is unrecognized. When these calls are removed the project builds and executes (except stalling in wasm_runtime_destroy unrelated to this).

### Replicate

Target: ESP-IDF ESP32
Build platform: Ubuntu

main.c (the main changes involve execting the wasm setup from a thread and loading the example module from a buffer in `module.h`)

```
#include <stdio.h>

#include "wasm_export.h"
#include "bh_platform.h"
#include "pthread.h"
#include "module.h"

#define THREAD_NUM 10

typedef struct ThreadArgs {
    wasm_exec_env_t exec_env;
    int start;
    int length;
} ThreadArgs;

void *
thread(void *arg)
{
    ThreadArgs *thread_arg = (ThreadArgs *)arg;
    wasm_exec_env_t exec_env = thread_arg->exec_env;
    wasm_module_inst_t module_inst = get_module_inst(exec_env);
    wasm_function_inst_t func;
    uint32 argv[2];

    if (!wasm_runtime_init_thread_env()) {
        printf("failed to initialize thread environment");
        return NULL;
    }

    func = wasm_runtime_lookup_function(module_inst, "sum", NULL);
    if (!func) {
        printf("failed to lookup function sum");
        wasm_runtime_destroy_thread_env();
        return NULL;
    }
    argv[0] = thread_arg->start;
    argv[1] = thread_arg->length;

    /* call the WASM function */
    if (!wasm_runtime_call_wasm(exec_env, func, 2, argv)) {
        printf("%s\n", wasm_runtime_get_exception(module_inst));
        wasm_runtime_destroy_thread_env();
        return NULL;
    }

    wasm_runtime_destroy_thread_env();
    return (void *)(uintptr_t)argv[0];
}

void *
wamr_thread_cb(wasm_exec_env_t exec_env, void *arg)
{
    ThreadArgs *thread_arg = (ThreadArgs *)arg;
    wasm_module_inst_t module_inst = get_module_inst(exec_env);
    wasm_function_inst_t func;
    uint32 argv[2];

    func = wasm_runtime_lookup_function(module_inst, "sum", NULL);
    if (!func) {
[... 236 lines cut ...]
```
[677/679] Linking CXX executable coop_demo.elf
FAILED: coop_demo.elf 
:
```
...

```
u newlib_include_heap_impl  -u newlib_include_syscalls_impl  -u newlib_include_pthread_impl  -u newlib_include_assert_impl  -Wl,--wrap=_Unwind_SetEnableExceptionFdeSorting  -Wl,--wrap=__register_frame_info_bases  -Wl,--wrap=__register_frame_info  -Wl,--wrap=__register_frame  -Wl,--wrap=__register_frame_info_table_bases  -Wl,--wrap=__register_frame_info_table  -Wl,--wrap=__register_frame_table  -Wl,--wrap=__deregister_frame_info_bases  -Wl,--wrap=__deregister_frame_info  -Wl,--wrap=_Unwind_Find_FDE  -Wl,--wrap=_Unwind_GetGR  -Wl,--wrap=_Unwind_GetCFA  -Wl,--wrap=_Unwind_GetIP  -Wl,--wrap=_Unwind_GetIPInfo  -Wl,--wrap=_Unwind_GetRegionStart  -Wl,--wrap=_Unwind_GetDataRelBase  -Wl,--wrap=_Unwind_GetTextRelBase  -Wl,--wrap=_Unwind_SetIP  -Wl,--wrap=_Unwind_SetGR  -Wl,--wrap=_Unwind_GetLanguageSpecificData  -Wl,--wrap=_Unwind_FindEnclosingFunction  -Wl,--wrap=_Unwind_Resume  -Wl,--wrap=_Unwind_RaiseException  -Wl,--wrap=_Unwind_DeleteException  -Wl,--wrap=_Unwind_ForcedUnwind  -Wl,--wrap=_Unwind_Resume_or_Rethrow  -Wl,--wrap=_Unwind_Backtrace  -Wl,--wrap=__cxa_call_unexpected  -Wl,--wrap=__gxx_personality_v0  -u __cxa_guard_dummy  -lstdc++  esp-idf/pthread/libpthread.a  -lgcc  esp-idf/cxx/libcxx.a  -u __cxx_fatal_exception  /home/amoeba/esp/esp-idf/components/xtensa/esp32/libxt_hal.a && :
/home/amoeba/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(main.c.obj):(.literal.wasm_start+0x54): undefined reference to `wasm_runtime_spawn_exec_env'
/home/amoeba/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(main.c.obj):(.literal.wasm_start+0x58): undefined reference to `wasm_runtime_destroy_spawned_exec_env'
/home/amoeba/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(main.c.obj):(.literal.wasm_start+0x5c): undefined reference to `wasm_runtime_spawn_thread'
/home/amoeba/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(main.c.obj):(.literal.wasm_start+0x60): undefined reference to `wasm_runtime_join_thread'
/home/amoeba/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(main.c.obj): in function `wasm_start':
/home/amoeba/Code/phd-wasm-experiments/spawn-thread/main/main.c:161: undefined reference to `wasm_runtime_spawn_exec_env'
/home/amoeba/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/amoeba/Code/phd-wasm-experiments/spawn-thread/main/main.c:180: undefined reference to `wasm_runtime_destroy_spawned_exec_env'
/home/amoeba/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/amoeba/Code/phd-wasm-experiments/spawn-thread/main/main.c:192: undefined reference to `wasm_runtime_destroy_spawned_exec_env'
/home/amoeba/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/amoeba/Code/phd-wasm-experiments/spawn-thread/main/main.c:206: undefined reference to `wasm_runtime_spawn_thread'
/home/amoeba/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/amoeba/Code/phd-wasm-experiments/spawn-thread/main/main.c:221: undefined reference to `wasm_runtime_join_thread'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

Any help would be appreciated, thank you!

TianlongLiang, 2023-06-09:
Hi, I will run this program and see what I can do, I will keep you updated :)

TianlongLiang, 2023-06-13:
Hi, @skrovk I tried your example program, it's a bug in CMakeLists.txt, set the cache variable before the `project` statement should be able to fix it, you can change your CMakeLists.txt to:
```CMake
cmake_minimum_required(VERSION 3.5)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

set (COMPONENTS ${IDF_TARGET} main freertos esptool_py wamr)
list(APPEND EXTRA_COMPONENT_DIRS "$ENV{WAMR_PATH}/build-scripts/esp-idf")

set (WAMR_BUILD_PLATFORM "esp-idf")
set (WAMR_BUILD_TARGET "XTENSA")

# set the cache variable WAMR_BUILD_LIB_PTHREAD will automatically set SHRAED_MEMORY, BULK_MEMORY and THREAD_MGR 
set (WAMR_BUILD_LIB_PTHREAD 1)

set (WAMR_BUILD_INTERP 1)
set (WAMR_BUILD_JIT 0)

project(wamr-simple)
```
Let me know if this solution works for you :)