wasm-micro-runtime/wasm-micro-runtime #2288: Wasm Loader: Double free error

arjunr2, 2023-06-12:
When loading the following compiled function in a module, I encounter a double free error. 
```(module
  (type (;0;) (func (param i32 i32)))
  (import "env" "memory" (memory (;0;) 2 1024 shared))
  (func $a_or (type 0) (param i32 i32)
    (local i32 i32 i32 i32 i32 i32 i32)
    global.get $__stack_pointer
    local.set 2
    i32.const 16
    local.set 3
    local.get 2
    local.get 3
    i32.sub
    local.set 4
    local.get 4
    local.get 0
    i32.store offset=12
    local.get 4
    local.get 1
    i32.store offset=8
    local.get 4
    i32.load offset=12
    local.set 5
    local.get 4
    i32.load offset=8
    local.set 6
    local.get 4
    local.get 6
    i32.store offset=4
    local.get 4
    i32.load offset=4
    local.set 7
    local.get 5
    local.get 7
    i32.atomic.rmw.or
    local.set 8
    local.get 4
    local.get 8
    i32.store
    return)
  (global $__stack_pointer (mut i32) (i32.const 71680))
  (export "memory" (memory 0))
)
```

TianlongLiang, 2023-06-13:
Hi, the wast file you posted seems incomplete, could you please upload the wasm file?

TianlongLiang, 2023-06-15:
Hi, [this pr](https://github.com/bytecodealliance/wasm-micro-runtime/pull/2295) should be able to fix this error. Thank you for spotting this issue, let me know if there are any other problems

closed 2023-06-21 by wenyongh