wasm-micro-runtime/wasm-micro-runtime #2297: Problems to support C++

xianjimli, 2023-06-18:
When developing app using C++, I encountered two problems:

* __cxa_atexit is not defined.
* Global objects are repeatedly constructed, where calling a wasm function once from native triggers construction once.
 
## Reproduce Steps:

Modify the basic sample by following these steps:

* Rename testapp.c to testapp.cpp and add the following code:
* change code to
```c++
/*
 * Copyright (C) 2019 Intel Corporation.  All rights reserved.
 * SPDX-License-Identifier: Apache-2.0 WITH LLVM-exception
 */

#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <stdint.h>

extern "C" {
int
intToStr(int x, char *str, int str_len, int digit);
int
get_pow(int x, int y);
int32_t
calculate_native(int32_t n, int32_t func1, int32_t func2);

//
// Primitive parameters functions
//
float
generate_float(int iteration, double seed1, float seed2)
{
    float ret;

    printf("calling into WASM function: %s\n", __FUNCTION__);

    for (int i = 0; i < iteration; i++) {
        ret += 1.0f / seed1 + seed2;
    }

    return ret;
}

// Converts a floating-point/double number to a string.
// intToStr() is implemented outside wasm app
void
float_to_string(float n, char *res, int res_size, int afterpoint)
{

    printf("calling into WASM function: %s\n", __FUNCTION__);

    // Extract integer part
    int ipart = (int)n;

    // Extract floating part
    float fpart = n - (float)ipart;

    // convert integer part to string
    int i = intToStr(ipart, res, res_size, 0);

    // check for display option after point
    if (afterpoint != 0) {
        res[i] = '.'; // add dot

        // Get the value of fraction part upto given no.
        // of points after dot. The third parameter
        // is needed to handle cases like 233.007
        fpart = fpart * get_pow(10, afterpoint);

        intToStr((int)fpart, res + i + 1, sizeof(res + i + 1), afterpoint);
    }
}

int32_t
mul7(int32_t n)
{
    printf("calling into WASM function: %s,", __FUNCTION__);
    n = n * 7;
    printf("    %s return %d \n", __FUNCTION__, n);
    return n;
}

int32_t
mul5(int32_t n)
{
    printf("calling into WASM function: %s,", __FUNCTION__);
    n = n * 5;
    printf("    %s return %d \n", __FUNCTION__, n);
    return n;
}

int32_t
calculate(int32_t n)
{
    printf("calling into WASM function: %s\n", __FUNCTION__);
    int32_t (*f1)(int32_t) = &mul5;
    int32_t (*f2)(int32_t) = &mul7;
    return calculate_native(n, (uint32_t)f1, (uint32_t)f2);
}
}

extern "C" int __cxa_atexit(void (*func)(void *), void *arg, void *dso_handle) {
  printf("__cxa_atexit:%p\n", func);
  return 0;
}

class FOO {
public:
  FOO() {
    printf("FOO\n");
  }
  ~FOO() {
    printf("~FOO\n");
  }
};

static FOO aFOO;

```
* 3. modify build.sh to build cpp

## Output log:

```
FOO
__cxa_atexit:0x3
calling into WASM function: generate_float
Native finished calling wasm function generate_float(), returned a float value: 102009.921875f
FOO
__cxa_atexit:0x3
calling into WASM function: float_to_string
calling into native function: intToStr
calling into native function: get_pow
calling into native function: intToStr
Native finished calling wasm function: float_to_string, returned a formatted string: 102009.921
FOO
__cxa_atexit:0x3
calling into WASM function: calculate
calling into native function: calculate_native, n=3, func1=1, func2=2
calling into WASM function: mul5,    mul5 return 15 
call func1 and return n1=15
calling into WASM function: mul7,    mul7 return 105 
call func2 and return n2=105
Native finished calling wasm function: calculate, return: 120
```

wenyongh, 2023-06-20:
Hi, the sample cpp file doesn't implement main function, and it seems intToStr/get_pow/calculate_native are implemented in native wrapper, and the build.sh isn't provided. Could you upload the whole sample?

xianjimli, 2023-06-21:
build.sh

```
#
# Copyright (C) 2019 Intel Corporation.  All rights reserved.
# SPDX-License-Identifier: Apache-2.0 WITH LLVM-exception
#

#!/bin/bash

CURR_DIR=$PWD
WAMR_DIR=${PWD}/../..
OUT_DIR=${PWD}/out

WASM_APPS=${PWD}/wasm-apps

rm -rf ${OUT_DIR}
mkdir ${OUT_DIR}
mkdir ${OUT_DIR}/wasm-apps

echo "#####################build basic project"
cd ${CURR_DIR}
mkdir -p cmake_build
cd cmake_build
cmake ..
make -j ${nproc}
if [ $? != 0 ];then
    echo "BUILD_FAIL basic exit as $?\n"
    exit 2
fi

cp -a basic ${OUT_DIR}

echo -e "\n"

echo "#####################build wasm apps"

cd ${WASM_APPS}

for i in `ls *.cpp`
do
APP_SRC="$i"
OUT_FILE=${i%.*}.wasm

# use WAMR SDK to build out the .wasm binary
/opt/wasi-sdk/bin/clang     \   
        --target=wasm32 -O0 -z stack-size=4096 -Wl,--initial-memory=65536 \
        --sysroot=${WAMR_DIR}/wamr-sdk/app/libc-builtin-sysroot  \
        -Wl,--allow-undefined-file=${WAMR_DIR}/wamr-sdk/app/libc-builtin-sysroot/share/defined-symbols.txt \
        -Wl,--strip-all,--no-entry -nostdlib \
        -Wl,--export=generate_float \
        -Wl,--export=float_to_string \
        -Wl,--export=calculate\
        -Wl,--allow-undefined \
        -o ${OUT_DIR}/wasm-apps/${OUT_FILE} ${APP_SRC}

if [ -f ${OUT_DIR}/wasm-apps/${OUT_FILE} ]; then
        echo "build ${OUT_FILE} success"
else
        echo "build ${OUT_FILE} fail"
fi
done
echo "####################build wasm apps done"
```

xianjimli, 2023-06-21:
> Hi, the sample cpp file doesn't implement main function, and it seems intToStr/get_pow/calculate_native are implemented in native wrapper, and the build.sh isn't provided. Could you upload the whole sample?

Yes, no main function. I need wasm export some functions and call them repeatedly.

wenyongh, 2023-06-25:
@xianjimli For the `Global objects are repeatedly constructed` issue, please add `-Wl,--export=__wasm_call_ctors` in build.sh when building wasm app.
For `__cxa_atexit` issue, I think currently it isn't supported by wasm spec, the spec doesn't specify which functions are to be called when the wasm instance is destroyed. My suggestion is to manually call the wasm functions before de-instantiate the wasm instance, or, before calling `wasm_runtime_deinstantiate`.

xianjimli, 2023-06-25:
OK, Thank you.

closed 2023-06-25 by xianjimli