wasm-micro-runtime/wasm-micro-runtime #2299: iwasm return Illegal instruction when executing a webassembly code

lyuxiaosu, 2023-06-20:
Prepare the following simple rust code:

```
fn main() {
    println!("Hello, world!");
}
```

Compile it to webassembly:
```
$ rustup target add wasm32-wasi
$ rustc hello.rs --target wasm32-wasi
```
Then use iwasm to execute it:

```
$ ./iwasm --version
iwasm 1.2.2
$ ./iwasm hello.wasm
Illegal instruction
```
Any suggestion for this? Thanks very much.

lyuxiaosu, 2023-06-20:
I build iwasm following the readme.txt

wenyongh, 2023-06-20:
@lyuxiaosu Which platform do you run iwasm? Could you try adding `cmake -DWAMR_DISABLE_WRITE_GS_BASE=1` when building iwasm?

lyuxiaosu, 2023-06-20:
> @lyuxiaosu Which platform do you run iwasm? Could you try adding `cmake -DWAMR_DISABLE_WRITE_GS_BASE=1` when building iwasm?

Hi wenyongh, thanks for the reply. Sorry I didn't clarify the platform. I built iwasm on x86_64, Ubuntu 20.04. 
I also tried the option you gave:

`cmake .. -DWAMR_BUILD_JIT=1 -DWAMR_BUILD_FAST_JIT=1 -DWAMR_DISABLE_WRITE_GS_BASE=1 -B build`

but it still showed the same error. Thanks.

wenyongh, 2023-06-20:
Hi, I tested it in my Ubuntu 20.04 x86_64: build hello.rs, build iwasm with `cmake .. -DWAMR_BUILD_JIT=1 -DWAMR_BUILD_FAST_JIT=1`, it runs OK.

Could you upload your wasm file?

lyuxiaosu, 2023-06-30:
I attached the wasm file, this file can be executed successfully with other wasm runtime, so it should be correct. The kernel version is `5.4.0-139-generic `, `ubuntu 20.04 x86_64`.
[hello.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/11919901/hello.zip)

lyuxiaosu, 2023-06-30:
I forgot to mention that I git clone the project and build iwasm. The git log shows:
```
commit 0ac5f206b81eea94e7c42b2358783c8cb79d56db (HEAD -> main, origin/main, origin/HEAD)
Author: Wenyong Huang <wenyong.huang@intel.com>
Date:   Fri Jun 16 11:48:12 2023 +0800

    Enable windows x86-32 AOT relocations (#2285)

    Implement relocation types IMAGE_REL_I386_DIR32 and IMAGE_REL_I386_REL32,
    fix failure to find AOT function symbol, and implement symbol __aulldiv.

```
I am not sure if it is the particular version caused the problem.

lyuxiaosu, 2023-06-30:
I git pull the latest code and rebuild everything and it still showed the same error:

```
xxx@node0:~/wasm-micro-runtime/product-mini/platforms/linux/build$ make clean
xxx@node0:~/wasm-micro-runtime/product-mini/platforms/linux/build$ cmake .. -DWAMR_BUILD_JIT=1 -DWAMR_BUILD_FAST_JIT=1
-- Fetching asmjit ..
** AsmJit Summary **
   ASMJIT_DIR=/users/xxx/wasm-micro-runtime/product-mini/platforms/linux/build/_deps/asmjit-src
   ASMJIT_TEST=FALSE
   ASMJIT_TARGET_TYPE=SHARED
   ASMJIT_DEPS=pthread;rt
   ASMJIT_LIBS=asmjit;pthread;rt
   ASMJIT_CFLAGS=
   ASMJIT_PRIVATE_CFLAGS=-Wall;-Wextra;-Wconversion;-fno-math-errno;-fno-threadsafe-statics;-fno-semantic-interposition
   ASMJIT_PRIVATE_CFLAGS_DBG=
   ASMJIT_PRIVATE_CFLAGS_REL=-O2;-fmerge-all-constants;-fno-enforce-eh-specs
-- Found LLVM 15.0.7
-- Using LLVMConfig.cmake in: /users/xxx/wasm-micro-runtime/product-mini/platforms/linux/../../../core/deps/llvm/build/lib/cmake/llvm
-- Build Configurations:
     Build as target X86_64
     CMAKE_BUILD_TYPE Release
     WAMR Interpreter enabled
     WAMR AOT enabled
     WAMR Fast JIT enabled with Lazy Compilation
     WAMR LLVM ORC JIT enabled with Lazy Compilation
     Multi-tier JIT enabled
     Libc builtin enabled
     Libc WASI enabled
     Fast interpreter disabled
     Multiple modules disabled
     Bulk memory feature enabled
     SIMD enabled
     Reference types disabled
-- Configuring done
-- Generating done
-- Build files have been written to: /users/xxx/wasm-micro-runtime/product-mini/platforms/linux/build
xxx@node0:~/wasm-micro-runtime/product-mini/platforms/linux/build$ make -j
Scanning dependencies of target vmlib
Scanning dependencies of target libiwasm
[  0%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/shared/platform/common/posix/posix_malloc.c.o
[  0%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/shared/platform/common/posix/posix_socket.c.o
[  0%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/shared/platform/common/posix/posix_thread.c.o
[  0%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/shared/platform/linux/platform_init.c.o
[  1%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/shared/platform/common/posix/posix_time.c.o
[  1%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/iwasm/common/wasm_exec_env.c.o
[  2%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/shared/platform/common/posix/posix_memmap.c.o
[  2%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/iwasm/common/wasm_runtime_common.c.o
[  2%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/random.c.o
[  2%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/shared/mem-alloc/ems/ems_hmu.c.o
[  3%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/iwasm/common/wasm_shared_memory.c.o
[  4%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/shared/mem-alloc/ems/ems_alloc.c.o
[  4%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/shared/mem-alloc/ems/ems_kfc.c.o
[  4%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/shared/mem-alloc/mem_alloc.c.o
[  4%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/shared/utils/bh_assert.c.o
[  4%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/shared/utils/bh_common.c.o
[  4%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/shared/utils/bh_log.c.o
[  6%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/shared/utils/bh_list.c.o
[  6%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/shared/utils/bh_hashmap.c.o
[  7%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/shared/utils/bh_vector.c.o
[  7%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/shared/utils/bh_queue.c.o
[  8%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/libc_wasi_wrapper.c.o
[  8%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/shared/utils/runtime_timer.c.o
[  8%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/iwasm/libraries/libc-builtin/libc_builtin_wrapper.c.o
[  8%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/posix.c.o
[  8%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/iwasm/common/wasm_application.c.o
[  9%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src/str.c.o
[  9%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/iwasm/common/wasm_native.c.o
[ 10%] Building C object CMakeFiles/libiwasm.dir/users/xxx/wasm-micro-runtime/core/shared/platform/common/posix/posix_malloc.c.o
[ 12%] Building C object CMakeFiles/vmlib.dir/users/xxx/wasm-micro-runtime/core/iwasm/common/wasm_memory.c.o
[ 12%] Building C object CMakeFiles/libiwasm.dir/users/xxx/wasm-micro-runtime/core/shared/platform/common/posix/posix_memmap.c.o
[... 231 lines cut ...]
[... cut ...]
[ 97%] Building C object CMakeFiles/iwasm.dir/main.c.o
[ 98%] Building C object CMakeFiles/iwasm.dir/users/xxx/wasm-micro-runtime/core/shared/utils/uncommon/bh_read_file.c.o
[ 98%] Building C object CMakeFiles/iwasm.dir/users/xxx/wasm-micro-runtime/core/shared/utils/uncommon/bh_getopt.c.o
[ 98%] Linking CXX executable iwasm
[100%] Linking CXX shared library libiwasm.so
[100%] Built target iwasm
[100%] Built target libiwasm
xxx@node0:~/wasm-micro-runtime/product-mini/platforms/linux/build$ ./iwasm ~/hello.wasm
Illegal instruction

```

wenyongh, 2023-08-08:
Hi, I tried building iwasm with the command as you mentioned `cmake .. -DWAMR_BUILD_JIT=1 -DWAMR_BUILD_FAST_JIT=1`, and run the `hello.wasm`, it works. Could you try using gdb to run, and when failed, use command like `disp /10i $pc` to see which instruction causes the core dump? And could you check whether interpreter mode works?